RobotLocomotion/drake #21872: Parameterize RgbdSensor in the Context

jwnimmer-tri, 2024-09-02:
**Is your feature request related to a problem? Please describe.**

One of our users (@rcory) has a randomized re-rendering pipeline that records a simulator trajectory, and then from that saved trajectory can re-render datasets with sensor images using altered scene details, e.g., changing the lighting, or environment map, or geometry textures, or etc.

Within the scope of things he aims to randomize, the _only_ attributes he wants to randomize that are not settable post-construction are the `RgbdSensor` constructor arguments (`parent_id`, `X_PB`, `color_camera`, `depth_camera`).  The main need is specifically for the `color_camera`, but we should aim to attack them all four elements with a uniform API approach, for consistency.

**Describe the solution you'd like**

The `RgbdSensor` system should offer a mechanism to adjust those values post-construction, either with setters that change the default value used during context-creation, or setters that take a mutable `Context` and change the parameters inside a context, or both.

Note that for `parent_id` and `X_PB`, there are some work-arounds: the MbP itself offers some ability to adjust the tree's kinematics post-construction, so by carefully choosing the sensor <=> plant attachment point, re-posturing is already possible.  This does not solve the `color_camera` problem, though.

**Describe alternatives you've considered**

Instead of adding new parameter API, the problem can already be solved within existing API by re-creating a new Diagram from scratch with the new values.  This is a more durable architecture overall and is how the re-rendering tool should have been written in the first place, but since the offline rendering tool has already hill-climbed into this local maxima, we'll plan to add this one missing API to unblock the work.

**Additional context**

See #20535 for an initial discussion of parameterizing X_PB.

jwnimmer-tri, 2024-09-02:
Other notes:

In our work on the MbT, we've come up with some API conventions / vocabulary for how to deal with configurable parameters.  We should consult that as an example, and probably try to align with it's spelling if we can.

We should probably do the "both" implementation option: offer setters without a context (to change the default) and with a context (to set inside a mutable context).  That is the most general tool to offer our users.  In general, I think such a choice should be Drake's nominal posture for all Parameters -- the user can always control both its default value, and its acute value in a given context.

I don't see any need to use _numeric_ (i.e., floating-point) parameters for any of the storage.  Everything can be abstract.  I don't think we'll ever do black-box gradient methods on these things, so we don't need any system-agnostic way to set their gradients.  I expect that storing everything as abstract will be the simplest code to write and test.

sherm1, 2024-09-03:
cc @sherm1

sherm1, 2024-09-03:
cc @sherm1

rpoyner-tri, 2024-09-05:
cc @rpoyner-tri I might take this if nobody else gets there first. No promise on when.

jwnimmer-tri, 2024-09-09:
Formally handing over to @rpoyner-tri to work on, later this week.

rpoyner-tri, 2024-09-12:
Are RgbdSensorAsync and SimRgbdSensor getting parallel treatment in this effort?

jwnimmer-tri, 2024-09-12:
`drake::systems::sensors::internalSimRgbdSensor` need not change.  It's not part of the Diagram.  The user can make a new one in case they need a new set of properties.  If implementing other parts of this feature need to change it then sure, but it's not part of the user story.

Yes the `RgbdSensorAsync` should change likewise, but probably that should be a second PR, since it's not on the critical path.

The `RgbdSensorDiscrete` is also relevant, but in that case I'd posit that the solution should be to teach the user how to change the parameters in the sub-Context by using `rgbd_sensor_discrete.sensor()` accessor to obtain the underlying `RgbdSensor` and then use the `RgbdSensor` API to make edits.  (In other words, _not_ to copy-paste the new setter functions onto `RgbdSensorDiscrete`.)

rpoyner-tri, 2025-01-17:
https://github.com/RobotLocomotion/drake/pull/22427 closes this out.