RobotLocomotion/drake #21920: Cannot close Schunk gripper to 0 position

hongkai-dai, 2024-09-17:
### What happened?

If the Schunk gripper is initialized to a non-zero position, and commanded to close the gripper to 0 position, then the gripper will just stay at the current position, as shown in my [branch](https://github.com/hongkai-dai/drake/tree/schunk_gripper_close_init). To reproduce this error, run the test
```
bazel test //manipulation/schunk_wsg:build_schunk_wsg_control_test
```
Note that if we command the gripper to close to 10mm, it works fine. Only 0mm position command causes problem.

The reason is that the context of the schunk gripper` context.get_discrete_state(0).last_target_position()` is initialized to 0. So if the target position is also 0, then in this if-else branch https://github.com/RobotLocomotion/drake/blob/b5c15082fe608f7f49fe779757ce53cdfe97148c/manipulation/schunk_wsg/schunk_wsg_trajectory_generator.cc#L91-L101
`last_traj_state->last_target_position() - target_position` = 0, and the code will not enter into the `if` branch that updates the trajectory, but jumps to the `else` branch. As a result, in `OutputTarget` function, since `trajectory_` is empty (because `UpdateTrajectory()` was not called), the schunk will just set the last position as the target position, as in https://github.com/RobotLocomotion/drake/blob/b5c15082fe608f7f49fe779757ce53cdfe97148c/manipulation/schunk_wsg/schunk_wsg_trajectory_generator.cc#L48-L55

One quick fix is to force calling `UpdateTrajectory()`. But I think a better solution is to default-initialize the context gripper `context.get_discrete_state(0).last_target_position()` to a non-zero value, like kInf?

cc @IanTheEngineer , this is why in the slack discussion https://tri-internal.slack.com/archives/C05N7QM7GGL/p1726093498365569 the gripper doesn't close at the beginning.

jwnimmer-tri, 2024-09-17:
Possibly easier -- if `trajectory_` is nullptr, then always recompute it?  Don't look at the history in that case.

sammy-tri, 2024-09-17:
>  if trajectory_ is nullptr, then always recompute it? Don't look at the history in that case.

I'd just go with this fix (and it works).  I can PR or if hongkai wants to add the change to his branch and PR that's fine too.
```
-  if (std::abs(last_traj_state->last_target_position() - target_position) >
-      kTargetEpsilon) {
+  if (!trajectory_ || (
+          std::abs(last_traj_state->last_target_position() - target_position) >
+          kTargetEpsilon)) {
```

PR #21925 (2024-09-18): [schunk_wsg] Fix trajectory bug with initial zero target
Update trajectory when trajectory_ is empty, so that the gripper can close its fingers at the beginning of the simulation.

resolves #21920

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21925)
closed 2024-09-18 by xuchenhan-tri