RobotLocomotion/drake #21924: Error with CalcPrincipalMomentsAndMaybeAxesOfInertia in Simple Example

cohnt, 2024-09-17:
### What happened?

I have a single obj model that I'm trying to load into the world, but I'm getting the following error message when I process the associated model directives:
```
    models = ProcessModelDirectives(directives, plant, parser)
RuntimeError: CalcPrincipalMomentsAndMaybeAxesOfInertia(): Unable to calculate the eigenvalues or eigenvectors of the 3x3 matrix associated with a RotationalInertia.
```
I tried printing out the matrix whose eigenvalues were being computed, and all entries were `nan`.

[This gist has a simple reproduction of the bug.](https://gist.github.com/cohnt/b791eecd186769dd5eb91a737faae394)

mitiguy, 2024-09-17:
FYI:  I checked the geometry in the file test.obj with one of my CAD programs.  
It displayed something strange -- picture attached.
It also displays NAN for inertia properties.
If the geometry is poorly formed, perhaps we could issue a better error message?
[image]

cohnt, 2024-09-18:
I've confirmed that fixing the direction of the faces no longer leads to an error. New `test.obj` attached below.

Having a better error message would be very helpful in this case -- I would have never thought that having faces in the wrong direction would lead to inertia errors.

```
g test

v 0.000000 0.000000 0.500000
v 0.000000 0.000000 -0.500000
v 0.000000 1.000000 0.500000
v 0.000000 1.000000 -0.500000
v 1.000000 0.000000 0.500000
v 1.000000 0.000000 -0.500000

f 1 5 3
f 2 4 6
f 1 3 2
f 2 3 4
f 3 5 4
f 4 5 6
f 5 1 6
f 6 1 2
```

mitiguy, 2024-09-19:
@cohnt  PR #21929 is meant to produce a better error message.
Feel free to provide guidance or comments here or there.
As of now, the error message is something like:
"CalcSpatialInertia(): The calculated volume of a triangle surface mesh is 0 whereas a reasonable positive value was expected. The mesh may have bad geometry, e.g., the winding (order of the vertices) of some faces do not produce outward normals."

cohnt, 2024-09-20:
Yes, that error message is very helpful. Thank you!

PR #21929 (2024-11-05): Improve exception message for bad geometry in .obj file.
This PR resolves issue #21924.  It creates a more coherent error message when the volume calculated from geometry in an .obj file is not reasonably positive.

Resolves #21924

### Note to release engineer: 
This PR is a <b>*fix*</b> to a broken contract.  This PR is <b>*not*</b> a breaking change.

Why?  Before this PR, the documentation in the file in geometry_spatial_inertia.h for the function 
```
SpatialInertia<double> CalcSpatialInertia(const geometry::TriangleSurfaceMesh<double>& mesh, double density);
```
included:
```
If these re [... cut]