RobotLocomotion/drake #21941: GCS AddRegions with edges_between_regions but without edge offsets defaults to zero offsets

sadraddini, 2024-09-23:
### What happened?

In [GcsTrajectoryOptimization::AddRegions ](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1trajectory__optimization_1_1_gcs_trajectory_optimization.html#a411eca99787867a84b5db6f5fd6c5cbc), I realized that when `edges_between_regions` is given but `edge_offsets` is `nullptr`, then the code assumes that the edge offsets are zero. 

It makes more sense that (I hope everyone agrees) that when `edge_offsets` is not present, it means the user does not have them (or too lazy to provide them) and the algorithm must compute them itself. 

cc @coh

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

binary tar.gz download

cohnt, 2024-09-23:
This was originally done for cases where the user knew the edges, but had no continuous revolute joints -- in that case, the need to carry around an appropriately-sized vector of zeros would be odd. Maybe the ideal fix is to implement this behavior for the current function call, and add an overload that only takes in `edges_between_regions`, to call the original function with the appropriately-sized `edge_offsets=[zeros]`? Since GcsTrajectoryOptimization is `@experimental`, presumably we don't need to worry about the breaking change.

sadraddini, 2024-09-23:
I see your point. 

The question is to how differentiate between the following: 

1. All the offsets are zero and the user does not want to bother creating an all zero appropriately sized vector. 
2. The user has edges but not the offsets (my case). They need to be computed. 

Your solution works but requires a subtle extra overload. No strong objection.

I still believe, as said before, that PairwiseIntersections should become its own class. Then we can have a constructor from a set of edges to fill all offsets as zero (can be a boolean flag) or leave them empty and as unknown.

cohnt, 2024-09-23:
I'll make a PR updating the behavior and adding the overload shortly.

> I still believe, as said before, that PairwiseIntersections should become its own class. Then we can have a constructor from a set of edges to fill all offsets as zero (can be a boolean flag) or leave them empty and as unknown.

I'm fine if you want to make that change, but I don't have the bandwidth to handle that right now.

PR #21946 (2024-10-01): Adjust GcsTrajectoryOptimization methods AddEdges and AddRegions to automatically compute edge offsets if they're not provided.
Resolves #21941.

I thought we would need an overload to handle the case where the user doesn't provide edge offsets because there are no continuous revolute joints, but actually, I don't think that's necessary. If there are no such joints, then no optimization problems will be solved, so the added computational load should be negligible.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21946)
closed 2024-10-01 by cohnt