RobotLocomotion/drake #21947: Reached Unreachable Code in BoxBoxGradient

cohnt, 2024-09-25:
### What happened?

While using SceneGraphCollisionChecker, I encountered the following error:
```
abort: Failure at geometry/proximity/distance_to_shape_touching.cc:228 in BoxBoxGradient(): condition 'Unreachable code was reached?!' failed.
```
I can provide code to replicate this, although it's a rather large chunk. I did print out the arguments to the function, in case that might help debug. But if that's insufficient, I can try to distill it down to a small runnable example.
```
box_A.side:	0.05, 0.55, 0.3
box_B.side:	0.25, 0.2, 0.15
X_WA	rpy = 0 -0 1.570000000000001 xyz = 1.2500000000000009 -2.9878953394393027e-16 0.15
X_WB	rpy = 3.141592653589793 -0 0 xyz = 1.25 0 0.15
p_ACa	1.71304e-17, 6.66134e-16, 8.32667e-17
p_BCb	2.22045e-16, 6.93889e-17, 8.32667e-17
```

### Version

4fdee8082686b878b97ab33d1605fddae6fc611d

### What operating system are you using?

Unknown / Other, Ubuntu 22.04

### What installation option are you using?

compiled from source code using CMake

jwnimmer-tri, 2024-09-26:
It's _possible_ that those numbers are enough to reproduce (or at least make a good guess), but ideally we would have the _exact_ numbers.  If you can re-run your repro and print all of the doubles without any truncation (i.e., full precision) that would help.  Also if you could print the rotations as the native 3x3 matrix (not rpy) that would also remove conversion noise.

In fact, once you have that, I imagine a C++ unit test call to `BoxBoxGradient` with your exactly numbers would probably immediately reproduce the problem.  It's just a namespace function with no state, so a full-precision argument dump should be enough to reproduce.

DamrongGuoy, 2024-09-26:
There's something fishy.  I can reproduce the problem with a unit test of `BoxBoxGradient`.  However, each of the given witness points p_ACa and p_BCb is almost at the center (0,0,0) of its box, far from the box's surface.  Witness points are supposed to be on the surface of the box.

The bug is likely at a higher level.  If @cohnt could give more data, I can check further.  For example, what went into `CalcDistanceFallback` [here](https://github.com/RobotLocomotion/drake/blob/da8ed366593e83daa093526ba2dedd988d7d5c31/geometry/proximity/distance_to_shape_callback.cc#L66).
```C++
template <>
void CalcDistanceFallback<double>(const fcl::CollisionObjectd& a,
                                  const math::RigidTransformd& X_WA,
                                  const fcl::CollisionObjectd& b,
                                  const math::RigidTransformd& X_WB,
                                  const fcl::DistanceRequestd& request,
                                  SignedDistancePair<double>* pair_data) {
```
From what already provided, I just need one more argument: the `const fcl::DistanceRequestd& request`.

Alternatively, distilling down to a small runnable example would be the best.

DamrongGuoy, 2024-09-27:
I reproduced the problem in [this commit](https://github.com/DamrongGuoy/drake/commit/ebde665cb6b80de3698d8a36b52ce90e849e58b7).  I'll continue on Monday.

cohnt, 2024-09-27:
Here are all of the values of the fields of `request`:

```
request.abs_err               	0
request.distance_tolerance    	1e-06
request.enable_nearest_points 	1
request.enable_signed_distance	1
request.gjk_solver_type       	0
request.rel_err               	0
```

Let me know if there are other values I can print for you! If you need a runnable example, I probably won't have time to make one until after October 3.

cohnt, 2026-10-27:
This was fixed upstream by flexible-collision-library/fcl#652, which Drake picked up in #24077 (first released in v1.50.0). I can no longer reproduce it on master.

The two boxes are concentric (centers differ by ~1e-15), so the Minkowski difference is symmetric about the origin and a face of GJK's tetrahedron passed exactly through the origin. The old `doSimplex4()` classified the origin's side of that face from rounding noise, so GJK cycled until `max_iterations` and reported the boxes as separated. FCL then returned a distance of ~3e-16 with both witness points at the shared center, and Drake treated that as touching and called `BoxBoxGradient()`. Building FCL standalone, the commit just before fcl#652 reproduces the bad result and fcl#652 gives the correct -0.1251.

closed 2026-10-27 by cohnt