RobotLocomotion/drake #21968: RLG fork of pybind11 incompatible with Python 3.13

jwnimmer-tri, 2024-09-28:
Error message from a CPython 3.13 wheel build:

```
pybind11/pytypes.h: In member function 'void pybind11::wrapper<Base>::resurrect_python3()':
pybind11/pytypes.h:2526:13: error: there are no arguments to '_PyGC_FINALIZED' that depend on a template parameter, so a declaration of '_PyGC_FINALIZED' must be available [-fpermissive]
2526 |         if (_PyGC_FINALIZED(patient_.ptr())) {
     |             ^~~~~~~~~~~~~~~
```

See https://bugs.python.org/issue40240 and https://bugs.python.org/issue32377.  The private macros `_PyGC_FINALIZED` and `_PyGC_SET_FINALIZED` are no longer available.  The former has a replacement function available, but the latter does not.  Those two macros are used by the bespoke RLG patch in our custom  `class wrapper`:

https://github.com/RobotLocomotion/pybind11/blob/51d715e037386fcdbeda75ffab15f02f8e4388d8/include/pybind11/pytypes.h#L2455-L2461

rpoyner-tri, 2025-01-06:
Three pieces of interest in the fork:
* unique_ptr args
* Wrapper
* eigen/numpy dtype magic for autodiff py objects

rpoyner-tri, 2025-01-06:
Above list is priority order for porting away. Jeremy has a pybind11 branch with partial reverts of the custom features.

jwnimmer-tri, 2025-01-07:
Here's my WIP branch to explore the problem: https://github.com/jwnimmer-tri/drake/commits/pybind11-tmp/

- It switches `pybind11` to my fork which is exactly an upstream release except for our `eigen/matrix.h` patch for dtypes.
- It cherry-picks some Drake PRs in flight.
- It puts `#if 0` around the remaining bindings that don't compile.

As such, everything _builds_ but some tests fail (due to the missing bindings).

The inventory if `#if 0` locations forms the worklist of problems that need solving.

Probably the next problem worth tackling is getting the `Simulator` working, since that halts a bunch of other tests so hides any deeper problems.

I have some small fixes for a couple of the other `#if 0` problems I'll also try to resurrect soon (e.g., LCM serializer bindings IIRC).

jwnimmer-tri, 2025-01-09:
See also https://github.com/RobotLocomotion/drake/issues/12811 for simulator-context discussion.

jwnimmer-tri, 2025-01-15:
FYI I've rebased my `-tmp` branch to cherry-pick all of the recent PRs (both open and merged PRs).

I'm going to start looking at the `ValueProducer::AllocateCallback` problems next (framework_py_semantics, framework_py_systems).

jwnimmer-tri, 2025-01-15:
Next I'm going to tackle the system scalar converter.

jwnimmer-tri, 2025-01-25:
I've rebased my pybind11-tmp branch atop everything in flight.  What still remains as `#if 0` are:
- AddMultibodyPlantSceneGraph
- CollisionChecker
- RandomSimulation / MonteCarloSimulation

jwnimmer-tri, 2025-01-27:
Also the final capstone -- we need to summarize the status quo / playbook for Drake Developers who maintain the bindings on an ongoing basis, probably at https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/pydrake_doxygen.h.

rpoyner-tri, 2025-02-06:
https://github.com/RobotLocomotion/drake/pull/22595 is in review.
Need to patch CollisionChecker, and that should be the end of unique pointer patching.

jwnimmer-tri, 2025-02-18:
@rpoyner-tri I rebased my https://github.com/jwnimmer-tri/drake/commits/pybind11-tmp/ WIP branch.  I see only the 7 test failures.

rpoyner-tri, 2025-02-26:
#22668 is merged, which takes care of one of the resistant failures.

I believe that the remaining problems are variations of https://github.com/pybind/pybind11/issues/1333 . It's not clear that there is a clean minimal patch available.

rpoyner-tri, 2025-02-28:
Ok, I've got a decently full sketch of solution, with some rationale still missing. I've got a full build to pass all tests using the following tagged branches:

https://github.com/rpoyner-tri/drake/releases/tag/pybind11-with-minimal-patches-tests-pass  -- at this tag, all tests pass when built against the matching pybind11 tagged branch. You'll need to patch the repository.bzl file, since it points to my local pybind11 checkout directory.

https://github.com/rpoyner-tri/pybind11/releases/tag/pybind11-with-minimal-patches-tests-pass -- this is the matching pybind11 branch. My intention is to take all the patches on this branch beyond the `v2.13.6` tag and turn them into drake-side patch files. Then we can build against v2.13.6 as the dependency. The patches will need to be carefully audited for proper co-auther attribution.

I'm about to disappear for a week, so feel free to experiment, etc.

rpoyner-tri, 2025-02-28:
https://github.com/RobotLocomotion/drake/pull/22685 is the last set of direct patches to pydrake that I can land independent of the switch-to-vanilla pybind patch set.

jwnimmer-tri, 2025-03-10:
@rpoyner-tri I started from the branches you listed above, removed the `local_repository_override` and pointed to your pybind11 branch online instead, and I see silent failure (probably segfault) from `//bindings/pydrake/planning:py/collision_checker_test FAILED in 0.8s`.  Then I also merged up to latest master and still see the same.  Here's what I did: https://github.com/jwnimmer-tri/drake/tree/python-313-wip.

Did I miss a step somewhere?  If it's still not ready that's fine, but if/when it is ready I want to toss it to kitware for #22581 preparation.

rpoyner-tri, 2025-03-10:
I have not seen the collision checker failure. I'm testing the pybind11 branch against newer drake master now, so I hope to see it.

rpoyner-tri, 2025-03-10:
I have the failure locally. Will investigate.

rpoyner-tri, 2025-03-10:
https://github.com/RobotLocomotion/drake/pull/22734

rpoyner-tri, 2025-03-14:
So, having completed the work we thought was necessary, I wanted some sort of independent check. I found https://github.com/pypa/abi3audit, which purports to scan .so's and complain about non ABI3 symbol use.

My experiment went like this:
```
$ pip install --user abi3audit
$ cd drake
$ bazel run //:install -- $(pwd)/dist
$ find dist -name '*.so' |xargs -ti abi3audit -v --assume-minimum-abi3 3.13 '{}' |& tee audit.log
```
The resulting log has a lot of complaints, but they may not actually block creation of a 3.13 wheel. For posterity, I will include the log output.

```
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/libdrake_marker.so
[13:44:07] 💁 libdrake_marker.so: 1 extensions scanned; 0 ABI version mismatches
           and 0 ABI violations found                                           
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/libdrake_lcm.so
[13:44:07] 💁 libdrake_lcm.so: 1 extensions scanned; 0 ABI version mismatches   
           and 0 ABI violations found                                           
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/python3.10/site-packages/lcm/_lcm.so
[13:44:07] 💁 _lcm.so: 1 extensions scanned; 0 ABI version mismatches and 0 ABI 
           violations found                                                     
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/python3.10/site-packages/pydrake/systems/framework.so
[13:44:08] 👎 : framework.so has non-ABI3 symbols                               
           ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┓                           
           ┃ Symbol                      ┃ Version  ┃                           
           ┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━┩                           
           │ PyMethod_Type               │ not ABI3 │                           
           │ PyStaticMethod_Type         │ not ABI3 │                           
           │ PyFrame_GetBack             │ not ABI3 │                           
           │ _PyThreadState_UncheckedGet │ not ABI3 │                           
           │ _PyObject_GetDictPtr        │ not ABI3 │                           
           │ PyInstanceMethod_New        │ not ABI3 │                           
           │ PyStaticMethod_New          │ not ABI3 │                           
           │ PyInstanceMethod_Type       │ not ABI3 │                           
           │ PyRun_StringFlags           │ not ABI3 │                           
           │ _PyType_Lookup              │ not ABI3 │                           
           └─────────────────────────────┴──────────┘                           
           💁 framework.so: 1 extensions scanned; 0 ABI version mismatches and  
           10 ABI violations found                                              
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/python3.10/site-packages/pydrake/systems/controllers.so
[13:44:08] 👎 : controllers.so has non-ABI3 symbols                             
           ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┓                           
           ┃ Symbol                      ┃ Version  ┃                           
           ┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━┩                           
           │ _PyType_Lookup              │ not ABI3 │                           
           │ PyMethod_Type               │ not ABI3 │                           
           │ PyInstanceMethod_New        │ not ABI3 │                           
           │ _PyThreadState_UncheckedGet │ not ABI3 │                           
           │ PyInstanceMethod_Type       │ not ABI3 │                           
           │ _PyObject_GetDictPtr        │ not ABI3 │                           
           │ PyFrame_GetBack             │ not ABI3 │                           
           └─────────────────────────────┴──────────┘                           
           💁 controllers.so: 1 extensions scanned; 0 ABI version mismatches and
           7 ABI violations found                                               
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/python3.10/site-packages/pydrake/systems/analysis.so
[13:44:08] 👎 : analysis.so has non-ABI3 symbols                                
           ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┓                           
           ┃ Symbol                      ┃ Version  ┃                           
           ┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━┩                           
           │ _PyType_Lookup              │ not ABI3 │                           
           │ _PyObject_GetDictPtr        │ not ABI3 │                           
           │ PyInstanceMethod_New        │ not ABI3 │                           
           │ PyInstanceMethod_Type       │ not ABI3 │                           
           │ PyRun_StringFlags           │ not ABI3 │                           
           │ PyFrame_GetBack             │ not ABI3 │                           
           │ PyMethod_Type               │ not ABI3 │                           
           │ _PyThreadState_UncheckedGet │ not ABI3 │                           
           └─────────────────────────────┴──────────┘                           
           💁 analysis.so: 1 extensions scanned; 0 ABI version mismatches and 8 
           ABI violations found                                                 
[... 414 lines cut ...]
           💁 __init__.so: 1 extensions scanned; 0 ABI version mismatches and 9 
           ABI violations found                                                 
[... cut ...]
           │ PyStaticMethod_New          │ not ABI3 │                           
           └─────────────────────────────┴──────────┘                           
           💁 perception.so: 1 extensions scanned; 0 ABI version mismatches and 
           10 ABI violations found                                              
abi3audit -v --assume-minimum-abi3 3.13 dist/lib/libdrake.so
[13:44:15] 💁 libdrake.so: 1 extensions scanned; 0 ABI version mismatches and 0 
           ABI violations found                                                 

```

jwnimmer-tri, 2025-03-14:
We are nowhere new ABI3 yet.  The 3.13 wheels will still use the traditional ABI.

closed 2025-03-14 by rpoyner-tri