RobotLocomotion/drake #21976: Add a function to scale Hyperellipsoids

Akshay5312, 2024-09-30:
There is built in functionality to scale HPolyhedrons (HPolyhedron::Scale) but not hyperellipsoids. I wished to find the approximate volume of a small high-dimensional polyhedron using the MaximumLargestInscribedEllipsoid method. The solver (not using an initial guess) was required to converge to an appropriate scale (which could be anywhere from ~1e-6 to 1e-20). Instead, it seemed more appropriate to scale the hpolyhedron by some factor, 'r', then scale the inscribed ellipsoid down by the same factor. before finding the volume. 

Similar to HPolyhedron::Scale(), a Hyperellipsoid::Scale(double 'r') might be useful. For every point x on the boundary of the original ellipsoid, there would be an associated x' = r*(x-center) + center on the boundary of the returned ellipsoid. This could simply be done by uniformly scaling 'A', resulting in Hyperellipsoid(A_ / r,  center_). 

While this could have been certainly done without the need for a specialized scaling method, I figured that it seemed a natural addition. I have linked a possible solution and test [here](https://github.com/RobotLocomotion/drake/compare/master...Akshay5312:drake:master).

hongkai-dai, 2024-10-02:
@Akshay5312 the code looks good, could you file a PR? Thanks!

hongkai-dai, 2024-10-02:
BTW, to stay consistent with `HPolyhedron::Scale()` function (which scales the h-polyhedron's volume by a factor of `scale`), `HyperEllipsoid::Scale()` should also scale by volume, namely the scaling factor on `A` should be `pow(scale, 1/ambient_dimension())` instead of `scale`.

Akshay5312, 2024-10-02:
I certainly can open a PR. 

As per 'scaling by volume', I believe that it may invalidate the original intent of scaling ellipsoids to manageable sizes in high dimensions. For ellipsoids with ambient_dimension=20, scaling up by a factor of 3 would require a scaling value of 3^20.

However, this seems like a very specific use case which I can implement locally on my computer and scaling by volume would definitely be more in line with the use of HPolyhedron::Scale. 

Should I continue with changing Hyperellipsoid::Scale to be by volume?

Akshay5312, 2024-10-02:
Opened PR #21985 with scale by volume

PR #21985 (2024-10-07): Add Hyperellipsoid::Scale() functionality
Closes #21976.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21985)
closed 2024-10-07 by xuchenhan-tri