RobotLocomotion/drake #21977: Investigate allowing additional Mac CI runners

BetsyMcPhail, 2024-10-01:
Now that Mac debug jobs have been removed (#21936) it may be possible to reduce the image size and allow more mac jobs to run at once.

The first step is to survey all of the current Mac jobs and determine how much disk space they are using.

nicolecheetham, 2024-10-02:
Investigative work on current Mac jobs (Note: Gi amounts may vary +- 1):
Maximum disk space used for a job: 46 Gi, 27% capacity
Occurs for 2 jobs: The mac-arm-ventura-clang-bazel-nightly-everything-release and the mac-arm-ventura-unprovisioned-clang-bazel-nightly-release

Sonoma Gi Usage:
1. 40 Gi, 24% capacity; 2 jobs 
        mac-arm-sonoma-clang-bazel-nightly-everything-release
        mac-arm-sonoma-unprovisioned-clang-bazel-nightly-release

2. 37 Gi, 22% capacity; 1 job
        mac-arm-sonoma-clang-bazel-continuous-release

3. 22 Gi, 13% capacity; 2 jobs
        mac-arm-sonoma-clang-cmake-nightly-everything-release
        mac-arm-sonoma-clang-cmake-nighlty-release

Ventura Jobs:
1. 46 Gi, 27% capacity; 2 jobs (already mentioned at the top)

2. 44 Gi, 26% capacity; 1 job
        mac-arm-ventura-clang-bazel-continuous-release

3. 29 Gi, 17% capacity; 4 jobs
        mac-arm-ventura-clang-cmake-nightly-everything-release
        mac-arm-ventura-clang-cmake-nightly-release
        mac-arm-ventura-unprovisioned-clang-cmake-nightly-packaging
        mac-arm-ventura-unprovisioned-clang-wheel-nightly-release

jwnimmer-tri, 2024-10-03:
Right now: 2 nodes, 2 VMs each.

New target: 2 nodes, 4 VMs each.

We might wait until we drop Ventura and make images for Sequoia before doing the switch-over.

williamjallen, 2024-11-01:
After further investigation, it looks like the current situation is 2 nodes, each of which can handle up to 1 VM at a time.

Each node has 16 GB of memory and 8 CPUs available.  Even though the storage bottleneck has been resolved, would splitting a node two ways create too much of a CPU or memory bottleneck?

jwnimmer-tri, 2024-11-01:
Ah, fair point.

So the question is whether we want to support 4 jobs running at once, each with half throughput -- or 2 jobs running at once, each at full throughput.

Depending on what kind of PR's are happening, either way could be the better answer, so let's just leave it alone for now.

closed 2024-11-01 by jwnimmer-tri