RobotLocomotion/drake #21982: Strange Behavior when Setting Light Intensity with Meshcat::SetProperty

cohnt, 2024-10-01:
### What happened?

If you run the following python code, the intensity of the ambient light is set to 2, as expected.
```
from pydrake.all import StartMeshcat
meshcat = StartMeshcat()
meshcat.SetProperty("/Lights/AmbientLight/<object>", "intensity", 2.0)
```
But if you instead call it without the decimal point, it sets the intensity to 1.
```
from pydrake.all import StartMeshcat
meshcat = StartMeshcat()
meshcat.SetProperty("/Lights/AmbientLight/<object>", "intensity", 2)
```

jwnimmer-tri, 2024-10-01:
The binding is overloaded to take the `value` as either `bool` or `float` or `str`.  I imagine that when given an integer, it is choosing `bool`.  The bool overload should probably be marked with `noconvert` so that it only accepts literal bools (https://pybind11.readthedocs.io/en/stable/advanced/functions.html#non-converting-arguments).

cohnt, 2024-10-02:
Good to know! Looks like [this guy here](https://github.com/RobotLocomotion/drake/blob/9339312b5eb6267b2916f2f9e3b88e92bef59a53/bindings/pydrake/geometry/geometry_py_visualizers.cc#L347)? I'll PR the fix later this week.

jwnimmer-tri, 2024-10-02:
Yes, that's _probably_ the overload where things are going wrong.  I would need to see some test case as final proof, but that would be the place I'd suggest starting.

PR #22021 (2024-10-12): Do not convert an int to a bool in Python binding of Meshcat::SetProperty
Previously, this lead to unintuitive behavior. For example, if you call `meshcat.SetProperty("/Lights/AmbientLight/<object>", "intensity", 2)`, then the light intensity is set to 1.

Closes #21982.

Since we don't have any sort of `meshcat::GetProperty` method, I don't think we have a way of testing this change?

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22021)
closed 2024-10-12 by jwnimmer-tri