RobotLocomotion/drake #22009: drake external examples: Mac drake_cmake_installed is failing

BetsyMcPhail, 2024-10-09:
Slack discussion: https://drakedevelopers.slack.com/archives/C270MN28G/p1728481302335769

Failing job: https://github.com/RobotLocomotion/drake-external-examples/actions/runs/11254479675/job/31291986238

BetsyMcPhail, 2024-10-09:
From the log (https://github.com/RobotLocomotion/drake-external-examples/actions/runs/11254479675):

```
macos sonoma 14 arm cmake installed
No such keg: /opt/homebrew/Cellar/tcl-tk

macos sonoma 14 arm cmake installed
No such keg: /opt/homebrew/Cellar/go

macos sonoma 14 arm cmake installed
No such keg: /opt/homebrew/Cellar/node@18

macos sonoma 14 arm cmake installed
Process completed with exit code 8.

macos sonoma 14 arm cmake installed
Cask go was renamed to homebrew/core/go.
```

As suggested by the comment (https://github.com/RobotLocomotion/drake-external-examples/blob/d6327fc16f7e4a008dbe922c6981c048f95a6dd2/.github/workflows/ci.yml#L39), it's likely time to remove the workarounds in ci.yml.

BetsyMcPhail, 2024-10-09:
Output from passing test (https://github.com/RobotLocomotion/drake-external-examples/actions/runs/11234931832/job/31231794157):

```
    Start 5: simple_bindings_test

5: Test command: /opt/homebrew/Frameworks/Python.framework/Versions/3.12/bin/python3.12 "/Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/src/simple_bindings/simple_bindings_test.py"
5: Working Directory: /Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/build/src/simple_bindings
5: Environment variables: 
5:  PYTHONPATH=/Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/build/src/simple_bindings:/opt/drake/lib/python3.12/site-packages
```

Output from failing test:

```
 Start 5: simple_bindings_test

5: Test command: /opt/homebrew/Frameworks/Python.framework/Versions/3.12/bin/python3.12 "/Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/src/simple_bindings/simple_bindings_test.py"
5: Working Directory: /Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/build/src/simple_bindings
5: Environment variables: 
5:  PYTHONPATH=/Users/runner/work/drake-external-examples/drake-external-examples/drake_cmake_installed/build/src/simple_bindings:/opt/drake/lib/python3.12/site-packages
```

jwnimmer-tri, 2024-10-14:
Fixed in https://github.com/RobotLocomotion/drake/pull/22012!

closed 2024-10-14 by jwnimmer-tri