RobotLocomotion/drake #22015: Un-inline virtual destructors

jwnimmer-tri, 2024-10-09:
Our style guide [outlaws](https://drake.mit.edu/styleguide/cppguide.html#Inline_Functions) inline virtual destructors, for the very good reason that leaving them inline sets ODR traps for their `type_info` weak objects.

In #22010, I cleaned up a few instances that users complained about, but we should circle back and clean all of them up.

As of my current checkout 5f766089, I see 115 instances we need to fix.  Here's the find command:

```
cd drake
find . -name examples -prune -o -name test -prune -o -name test_utilities -prune -o -name '*.h' -exec grep --color -nHE --null -e '(virt.*~.*\(\).*[={]|*~.*\(\).*(over|final).*[={])' \{\} +
```

By directory, that is:
```
     14 ./systems/analysis
     13 ./systems/framework
     12 ./multibody/inverse_kinematics
      8 ./multibody/optimization
      5 ./systems/primitives
      5 ./solvers
      4 ./planning/trajectory_optimization
      4 ./geometry/optimization
      3 ./multibody/contact_solvers
... etc ...
```

jwnimmer-tri, 2025-03-28:
New report as of today:

```
     12 ./multibody/inverse_kinematics
      8 ./multibody/optimization
      7 ./systems/framework
      3 ./multibody/contact_solvers
      2 ./multibody/fem
      2 ./geometry/optimization/dev
      1 ./systems/lcm
      1 ./systems/controllers
      1 ./multibody/triangle_quadrature
      1 ./multibody/parsing
      1 ./multibody/benchmarks/chaotic_babyboot/MG
      1 ./common
```

jwnimmer-tri, 2025-11-06:
> ... then the linker combines the equivalent type_info objects ...

Not necessarily.  If there are multiple `*.so` files that all contain the `type_info` (e.g., in our Python bindings), they might not be combined.  That's why we aim to emit just one copy of the `type_info`, so its uniqueness is guaranteed. We build with `--copt=-fno-assume-unique-vtables` to help avoid that trap, but not everyone does.

It's also simply wasteful to emit the vtable (and all of the virtual methods!) into every translation unit, and then let the linker throw away most of them.  Like all other inline methods, the multiple emitted copies bloats the build.  Virtual destructors are just more bloat than most cases since they are usually the first virtual in a class which means it's the "key function" that emits the vtable and everything _it_ needs.

j4yyousi, 2025-11-08:
> Virtual destructors ... are usually the first virtual in a class which means it's the "key function" that emits the vtable and everything it needs.

I see, so the vtable and `type_info` for a class are emitted in the translation unit that defines the first non-inline virtual function (usually the destructor).

j4yyousi, 2025-11-14:
Hi, should my last PR close this issue? Since the four remaining instances should be ignored:

1. `./systems/lcm/serializer.h61: ~Serializer() override = default;` is an open-ended template
2. `./multibody/benchmarks/chaotic_babyboot/MG/MG_chaotic_babyboot_auto_generated.h35: virtual ~MGIntegrator() {};` is in a generated file that should not be edited.
3. `./multibody/fem/force_density_field_base.h38: virtual ~ForceDensityFieldBase() = 0;` destructor already defined in `drake/multibody/fem/force_density_field_base.c`
4. `./common/value.h243: ~Value() override = default;` is an open-ended template

closed 2025-11-15 by jwnimmer-tri