RobotLocomotion/drake #22019: Send GHA failure notifications to build cop channel automatically

williamjallen, 2024-10-10:
Those who are subscribed to the build cop mailing list get emails about failing Jenkins jobs, but not for failing GitHub Actions jobs.  These GitHub Actions failures can be reported to the build cop channel automatically via the official GitHub+Slack [integration](https://github.com/integrations/slack).  If desired, something [similar](https://www.jenkins.io/doc/pipeline/steps/slack/) could be done for Jenkins as well.

jwnimmer-tri, 2024-10-10:
I will defer the importance / priority of this to the actual buildcops (of which I'm not one).  As a non-cop, I just check https://drake-jenkins.csail.mit.edu/view/Production/ every morning to survey our health.  For buildcops who are supposed to triage more quickly than daily, I can see how a slack ping would help.

jwnimmer-tri, 2025-04-24:
On the main page https://drake-jenkins.csail.mit.edu/ I see the little green DEE box.  I don't see it on the https://drake-jenkins.csail.mit.edu/view/Production/ page, but I thought it used to be there?

If there is an easy way to put it on the https://drake-jenkins.csail.mit.edu/view/Production/ page, that might be a good compromise; having the cops check https://drake-jenkins.csail.mit.edu/view/Production/ daily is a good idea in any case -- it's a good dashboard for them.

tyler-yankee, 2025-04-24:
I'm not sure if it used to be there or not, but either way I just added it back. (Might have been lost when we updated in https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/160. We have to add that view by hand, since the job builder doesn't support the dashboard/grid.)

tyler-yankee, 2025-04-24:
The buildcop docs reference the Continuous and Nightly Production subset views a lot, rather than the big Production view. Perhaps we add DEE's GHA to either or both of those (it's definitely nightly, and it _is_ "continuous," but to DEE rather than Drake itself), and then I'm happy to call this closed.