RobotLocomotion/drake #22026: Investigate major version upgrade of "Badge" Jenkins plugin

williamjallen, 2024-10-11:
The "Badge" Jenkins plugin has released a new major version which is not backwards compatible with the currently installed version.  This issue serves as a reminder that we should do further investigation before upgrading.

[image]

BetsyMcPhail, 2025-01-23:
This is a duplicate of #22476.

williamjallen, 2025-01-24:
@BetsyMcPhail #22476 is for a separate major version upgrade.

tyler-yankee, 2025-03-13:
Okay, looks like this plugin saw a complete rework in version 2.x. Notes on breaking changes that don't affect us, but I'd still like to note/document here:

* They note some downstream incompatibility with other plugins, that as far as I'm aware we don't use (Groovy Postbuild and Build Monitor).
*  Some changes to global configuration if you use Jenkins Configuration as Code (JCasC), which we don't.
* Some changes to how HTML is rendered in badges, which shouldn't apply to us.

From what I can tell, the only thing we use this plugin for is to put the CDash icon with link next to the builds in the left-hand side menu, and put a similar icon/link in the build summary. The relevant code is at the end of each pipeline script:
```
addBadge icon: '/userContent/cdash.png', link: cDashUrl, text: 'View in CDash'
createSummary icon: '/userContent/cdash.png', text: "<a href=\"${cDashUrl}\">View in CDash</a>"
```

*`createSummary` is deprecated in 2.x and replaced by `addSummary`.* This means the relevant code in drake-jenkins-jobs will need to change and be redeployed, probably after the [major version upgrade of JJB lands](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/160). We'll just need to be careful about the order of operations of the deployment and plugin update (I'm not sure how that should go, actually?). 

The release notes for 2.0 also talk about `text` and `link` being "new fields" of `addBadge`, which I don't understand, because we're clearly using them with 1.13.

The other minor releases since 2.0 don't appear to have anything significant to worry about.

tyler-yankee, 2025-03-13:
I should have read more carefully. On `createSummary`:

> Remains backwards compatibility for existing pipelines and configs

I think we should still make the change to `addSummary` for future-proofing if we have time, but it takes the pressure off if we'd like to update the plugin now.

williamjallen, 2025-03-13:
While we're tinkering with this plugin, it would be great to update the CDash logo to the [most recent](https://github.com/Kitware/CDash/blob/master/public/img/cdash.svg) one.  It would also probably be useful to link to builds directly rather than linking to filtered results which show a single build.  I can do those things in follow-up work if it's not directly related to this update.

tyler-yankee, 2025-03-13:
Yeah, I'd support both of those ideas in follow-up work once we know things are stable after the update. 

For the logo update, either:
* That new file needs to be stored on the drake-jenkins server, since it's currently pulling the file from local storage; or
* The current code could be updated to use a [link](https://www.jenkins.io/doc/pipeline/steps/badge/#addbadge-add-badge) as the image source, which perhaps keeps things more current for future changes -- assuming that link remains the same.

tyler-yankee, 2025-03-17:
Successfully updated; still waiting on [#162](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/162) to be merged for us to move to the new functions.

tyler-yankee, 2025-03-26:
Closed as completed via [#162](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/162).

closed 2025-03-26 by tyler-yankee