RobotLocomotion/drake #22038: issue with bazelisk_1.22.0 -amd64.deb

Dawitsco, 2024-10-15:
### What happened?

/tmp/bazelisk_1.22.0-amd64.deb: OK

(Reading database ... 352591 files and directories currently installed.)
Preparing to unpack /tmp/bazelisk_1.22.0-amd64.deb ...
Unpacking bazelisk (1.22.0) ...
dpkg: error processing archive /tmp/bazelisk_1.22.0-amd64.deb (--install):
 trying to overwrite '/usr/bin/bazel', which is also in package bazel-bootstrap 3.5.1+ds-3ubuntu1
Errors were encountered while processing:
 /tmp/bazelisk_1.22.0-amd64.deb
The Drake source distribution prerequisite setup script has experienced an error on line 20 while running the command source "${BASH_SOURCE%/*}/source_distribution/install_prereqs.sh" "${source_distribution_args[@]}"

Bazel version:
Build label: 3.5.1- (@non-git)
Build target: bazel-out/k8-dbg/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Sun Aug 15 22:54:48 2021 (1629068088)
Build timestamp: 1629068088
Build timestamp as int: 1629068088

cmake:
cmake version 3.22.1

Dawitsco, 2024-10-15:
this operation was done on ubuntun 22.04.05 LTS

jwnimmer-tri, 2024-10-15:
Yes, the `bazelisk` package is incompatible with `bazel-bootstrap`.  Please try running `install_prereqs --without-bazel`, instead.  You might also need to `sudo apt remove bazelisk` or similar to recover, first.

I should probably change the default to *not* install bazelisk when using CMake for the build.

Dawitsco, 2024-10-15:
Thanks for your response. i actually used Bazel build, do you mean i should install it directly without the build?

Dawitsco, 2024-10-15:
if you don't mind, can you explain the installation process to do this.Thanks

Dawitsco, 2024-10-15:
haha i think i got it to work but got this after running it:
INFO: Reading rc options for 'version' from /home/dawitsco/drake_C/drake/tools/bazel.rc:
  Inherited 'common' options: --enable_workspace=true --enable_bzlmod=false
ERROR: --enable_workspace=true :: Unrecognized option: --enable_workspace=true
The Drake source distribution prerequisite setup script has experienced an error on line 20 while running the command sudo -u "${SUDO_USER}" bash "${user_env_script}

closed 2024-10-15 by Dawitsco