RobotLocomotion/drake #22040: Use venv for macOS when installation method is tgz

jwnimmer-tri, 2024-10-15:
**Is your feature request related to a problem? Please describe.**

When installing a binary release of Drake on macOS using the `tar.gz` download installation method (aka [from_binary](https://drake.mit.edu/from_binary.html)), Drake currently runs a pip command:

https://github.com/RobotLocomotion/drake/blob/82999cb1c76d43d4493afd5ff39b840a0d4d65b8/setup/mac/binary_distribution/install_prereqs.sh#L70

Note that we are running the `pip3.12` command from homebrew, which means that `--break-system-packages` is installing Drake's python dependencies system-wide.  That is not a good idea for just to run software library someone just downloaded.  They should be able to run Drake without polluting their system-wide homebrew python path.

We should ensure that users always run using some kind of virtual environment, and provide good docs/code in support of that.

Links to topical end-user documentation:
- https://drake.mit.edu/from_binary.html#use-as-a-python-library
- https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed#readme

**Describe the solution you'd like**

(1) Change anything and everything that's necessary (Drake's binary packages, Drake's install_prereqs scripts, Drake's documentation, DEE's code, DEE's documentation, etc) so that running a Drake tgz binary release on macOS does not pollute homebrew python.

(2) The `binary_distribution/Brewfile` must no longer cite `numpy`, but everything else there (`gcc`, `python@3.12`, etc.) is satisfactory to leave alone.  We still want to use Homebrew's CPython interpreter, just not any of its third-party Python libraries.  Drake should always be running inside of a virtual environment.

(3) The `binary_distribution/requirements.txt` must not be installed system-wide, but rather inside of a virtual environment.  The documentation at https://drake.mit.edu/from_binary.html#use-as-a-python-library already has a start on this, but isn't well-matched for macOS because `python3` might not point to homebrew python.  Simply changing the documentation would be a plausible solution, but I imagine that combining these two lines ...
```sh
python3 -m venv env --system-site-packages
env/share/drake/setup/install_prereqs
```
... into just a one-liner call to `env/share/drake/setup/install_prereqs` might be less brittle (so that the python details can be scripted, instead of copy-pasted from the docs).

(4) Any scripts that we provide should (as much as practical) give the user good error messages in case they make a mistake, e.g., accidentally using Apple Python, or accidentally trying to run `pip` into the `/opt/homebrew` tree, etc.

(5) Fix the documentation at https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed to promise support for macOS (and by implication then also operate correctly on both Ubuntu and macOS).

**Describe alternatives you've considered**

n/a

**Additional context**

This is a sub-task of the #8392 epic.

jwnimmer-tri, 2024-12-28:
> Activating the venv is always going to be an issue.

That's a good task for our documentation.  The documentation should either (or both) teach the user to activate the environment, or explain how to run `venv/bin/python3` as their interpreter without the need to activate.  As noted in the OP, we already have some documentation to that effect (https://drake.mit.edu/from_binary.html#use-as-a-python-library).

Or were you saying that our code would somehow need to self-activate for some reason, automatically?  I don't understand why that would be the case, so please explain if so.

> In any case, I think we either need to decide where the venv should live (probably within the Drake install), or just require an active venv to run install_prereqs. WDYT?

We already recommend a specific layout for Python users:
https://drake.mit.edu/from_binary.html#use-as-a-python-library

I think my default answer would be to keep the layout the same when extending the instruction to all users, unless we have some specific reason why some change would be either required, or at least an improvement.  I could possibly see renaming `env` to `drake` as possibly being more clear or straightforward for users?

See also note (3) in the OP, which is related.

Another factor in what kind of layout works correctly would be how the CMake recipe in (5) goes about its business.

mwoehlke-kitware, 2024-12-31:
> We already recommend a specific layout for Python users: https://drake.mit.edu/from_binary.html#use-as-a-python-library

That seems to lean toward 'require an active venv to run install_prereqs'...

jwnimmer-tri, 2024-12-31:
Some people (myself included) will refuse to ever _activate_ a venv.  Activation means setting shell variables that are sticky, which leads to difficult-to-replicate state from unknown past changes when reproducing a problem.  My shells never have any state -- I will always spell out the full path to e.g. `venv/bin/python3 -m some_module` when I want to run something in the context of some venv.

I want to make sure our users also have the ability to not activate, if they don't want to.  We must not strictly _require_ an activated venv.  It's fine to _recommend_ activating if that is helpful, but it must not be absolutely required.

The example command on our install page currently says:
> Install dependencies within the environment:
> `env/share/drake/setup/install_prereqs`

Under that regime, it seems to me like `install_prereqs` could default to using `$(dirname $0)/../../../../bin/python3` to install its dependencies, without requiring activation?

But I still think that's missing the forest for the trees.

Please read again more carefully what I wrote in (3) in the OP.  I took time to plan this out and it doesn't seem like it's registered with you yet, or else my proposal has some problem which I don't understand yet.  I think `install_prereqs` can make the venv itself, from scratch.  I'd think the ideal install documentation for a binary tgz would be something like:

```
tar -xvzf drake.tar.gz
drake/share/drake/setup/install_prereqs
```

If we can give the user a one-liner to run (just `install_prereqs`) and that one script does all of right things automatically, surely that is easier than trying to teach users how to do multiple independent steps with a venv?

After that one-liner, something like `drake/bin/meldis` should work automatically (even without activation). We can document that users who want to run Drake interactively could run `drake/bin/python3` to get all of the paths they need, or they could also choose the active the environment if they wish.

mwoehlke-kitware, 2025-01-03:
> We must not strictly require an activated venv. It's fine to recommend activating if that is helpful, but it must not be absolutely required.

...But this goes back to my original point; `install_prereqs` needs to know where the venv lives. (And, no, I don't want to require a user-provided venv to be in a specific location relative to `install_prereqs`.) That means either `install_prereqs` *creates* the venv in the first place (and we need to be sure we're adequately communicating to users how to "do stuff" so that the Drake-managed venv will be available), or needs to be pointed at it. The easiest / most reliable mechanism for the latter is to require that the venv is active, i.e. `VIRTUAL_ENV` is set.

>  I think `install_prereqs` can make the venv itself, from scratch.

...to which I replied "we ... need to decide where the venv should live", but, rather than respond to that, your reply seemed to lean toward the user supplying the venv.

So, I will ask again: *where should this Drake-managed venv be rooted?* Your mention of `drake/bin/python3` suggests "the same place as `drake.tar.gz` was extracted"?

jwnimmer-tri, 2025-01-03:
The other thing you could do to help establish the requirements is to work out how https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed will interact with the venv.

mwoehlke-kitware, 2025-01-03:
> The other thing you could do to help establish the requirements is to work out how https://github.com/RobotLocomotion/drake-external-examples/tree/main/drake_cmake_installed will interact with the venv.

It... won't? Aren't those users not Python users?

jwnimmer-tri, 2025-01-03:
That example does use Python.  See `src/find_resource/find_resource_example.py` and `src/simple_bindings/...`.

mwoehlke-kitware, 2025-01-03:
Well, "These instructions are only supported for Ubuntu". 🙂

Uh... am I missing something, or do those instructions not ever run Drake's `install_prereqs`? Above tongue-in-cheek notwithstanding, how *do* those instructions work on macOS today?

jwnimmer-tri, 2025-01-03:
> "These instructions are only supported for Ubuntu".

Yup.  Thus the call to action in the OP: "(5) Fix the documentation ... to promise support for macOS."

> ... how do those instructions work on macOS today?

Check the `drake_cmake_installed/.github/**` files.  There is a script `drake_cmake_installed/setup/install_prereqs` which shells out to the installed copy of the prereqs script (after installing beyond-Drake dependencies used by the example code, e.g., gtest).

In the README that's mentioned as the first command in the shell snippet:

https://github.com/RobotLocomotion/drake-external-examples/blob/3e1ba134bcbd2210b4827476c2eaadb9fc91bf20/drake_cmake_installed/README.md#L13-L14

mwoehlke-kitware, 2025-01-06:
DEE currently does this (to make Drake itself available):
```cmake
get_filename_component(PYTHONPATH
  "${drake_DIR}/../../python${Python3_VERSION_MAJOR}.${Python3_VERSION_MINOR}/site-packages"
  REALPATH
)
```

If the venv is rooted at Drake's install, I _think_ that will put the venv packages in the same place, and we may not need to change anything?

mwoehlke-kitware, 2025-01-28:
>  Is that still on @mwoehlke-kitware's plate?

I'm happy for it to not be. (Feel free to tag me on reviews, though.)

BetsyMcPhail, 2025-01-28:
@tyler-yankee can you please coordinate with @mwoehlke-kitware to finish this last item?

PR #22607 (2025-02-10): Use separate venv for installation PDM
Modify how we set up the virtual environment for macOS installs to keep PDM in a separate environment. This avoids PDM modifying its own environment (which appears to have possible race conditions).

Amends #22477.
Closes #22040.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22607)
closed 2025-02-10 by jwnimmer-tri