RobotLocomotion/drake #22042: Wheel build should use a locked requirements file

jwnimmer-tri, 2024-10-15:
**Is your feature request related to a problem? Please describe.**

In our wheel build, currently we install some unpinned dependencies:

https://github.com/RobotLocomotion/drake/blob/82999cb1c76d43d4493afd5ff39b840a0d4d65b8/tools/wheel/macos/build-wheel.sh#L112-L115

https://github.com/RobotLocomotion/drake/blob/82999cb1c76d43d4493afd5ff39b840a0d4d65b8/tools/wheel/image/provision-python.sh#L41-L54

This is a problem because it means our wheel builds are not reproducible (so checking out a Drake tag and building it might produce a different result, or even not work at all), and that broken uploads to PyPI can break our Nightly CI out from under us.

**Describe the solution you'd like**

- Compute requirements lockfile(s) for our wheel builds, so that the versions of everything we need are consistent across repeated builds of the same Drake source tree.
- Ensure that the lockfile is regularly refreshed (ala #22041).

**Describe alternatives you've considered**

n/a

**Additional context**

At the moment, in non-wheel builds, we only have venv lockfiles for macOS.  This issue should lock both the Linux wheel build and the macOS wheel build, i.e., the two code snippets above.

jwnimmer-tri, 2024-12-16:
From f2f, implementation options:

(1) Run the jazzband lockfile generator a half dozen times (or whatever) on the matrix of python versions / platforms, to have N different lockfiles.

(2) Find a lockfile creation tool that can resolve dependencies in an interpreter and platform agnostic way, have just one lockfile.

(3) Just type in hard-coded numbers in https://github.com/RobotLocomotion/drake/blob/master/tools/wheel/image/provision-python.sh, without a lockfile, and without any automated re-locking tool

Doing option (2) would be best if we can find and wrangle such a tool.  Otherwise we'll fall back to (3) and live with it indefinitely.

mwoehlke-kitware, 2024-12-26:
PDM looks plausible for (2) (see also #22349), and if/when that lands, it shouldn't be a big deal to use it for the *macOS* wheels. However, having a separate implementation for the Linux wheels might be messy. IMHO, if we go in that direction, it would be best to tackle #22040 first, then locking for macOS wheels *only*, then switch *Linux* non-wheel builds over to using PDM for Python dependencies, and only lock the Linux wheel-build dependencies once PDM is already in place otherwise.

jwnimmer-tri, 2024-12-28:
Sure, I think it's fine to sequence the work as "first replace Jazzband with something better (e.g., PDM)", before we add more uses of locking.

We can revisit the question of next steps w.r.t. this issue after that part is finished.  I anticipate that it should be plausible (and preferable) to lock the manylinux wheel builds on their own, without messing with the Ubuntu builds.

PR #22505 (2025-01-27): Use @python to provide Linux wheel dependencies
Use the managed virtual environment to provide Python dependencies for Linux wheel builds. This allows Linux wheels to version-pin their dependencies the same way we're already doing for macOS.

This includes several additional changes:

- Modify `venv_sync` to make the virtual environment available in a known location for wheel builds.

- Use `patchelf` from Ubuntu's repositories rather than PyPI. We originally needed a newer version to avoid problems with auditwheel (n.b. https://github.com/pypa/auditwheel/issues/103), but this is no longer nec [... cut]