RobotLocomotion/drake #22045: Girder shutdown

jwnimmer-tri, 2024-10-16:
It's time to shut down https://drake-girder.csail.mit.edu/.  We aren't using it anymore, so we should save the money (and maintenance cost).

Let's do it in phases:

(1) When next we have some free time, we should turn off the infrastructure -- but not delete anything.

(2) Then maybe 3 months later, we'll actually go and delete (or glacier?) the S3 bucket, delete all of the server images and/or configs, whatever makes sense.

BetsyMcPhail, 2025-02-13:
The initial clean-up pass is done. I set a reminder to myself to check on the backup cleanups in about 5 weeks. In the meantime, moving this to "To Do"

jwnimmer-tri, 2025-02-18:
> @jwnimmer-tri do you have an example or two of stored files we may want to find. I think I found them, but want to be sure.

There's a file `example.txt` with sha512 of `2cf9faaa45a706a84048424b47622dfe0d98496b87370bb5278eaaf5a7f17cb9f382781b6782321e16304f55b0cfcfb9f8d6ce23aedd6e939313124efa8168ad`.

There's a file `artist_workshop_4k.hdr` with a sha512 of `a4c534b81e2730bc35a24c452d3498422eb773ba8d4d396d0d5b981414eb341c2d4ee3295e752b4ce592e644e0d40ab0ed5045f11212600329bd82188d235b03`.

BetsyMcPhail, 2025-02-24:
Surprisingly, I can find `example.txt` but not `artist_workshop_4k.hdr`. I'm fairly confident that all of the data that is available is in the `drake-girder` bucket, I don't see any other bucket that would make sense. There are about 8600 files in the bucket, the latest added in Sept. 2024. The drake-mongo-data volume isn't large enough to be storing data (it's 100Gi whereas there's ~350GB of files in the bucket).

jwnimmer-tri, 2025-02-26:
My mistake, the hdr file was added October 2024, after the shutdown.  I think we're safe to call this done.

jwnimmer-tri, 2025-03-04:
One last bit of clean-up to consider: should we drop the CSAIL DNS records that we no longer need?  Possibly that would be `drake-mongo` and `drake-webdav` and `drake-girder`.

BetsyMcPhail, 2025-03-10:
> One last bit of clean-up to consider: should we drop the CSAIL DNS records that we no longer need? Possibly that would be `drake-mongo` and `drake-webdav` and `drake-girder`.
> 
> Edit: Done already as part of resolving [#22694](https://github.com/RobotLocomotion/drake/issues/22694

Note that `drake-webdav` is the Linux cache server. As far as I know, it is never accessed via `drake-webdav.csail.mit.edu` so the decommissioning should be fine.

BetsyMcPhail, 2025-04-21:
The remaining step is to delete or move the girder buckets to glacier.

drake-girder ~349 GB
elasticbeanstalk-us-east-1-333343588315 ~14GB 

Current storage cost $0.023 / GB (~$8.35/month)
Glacier storage cost $0.0036 / GB (~$1.31/month)

BetsyMcPhail, 2025-07-08:
The girder S3 buckets have been deleted.

That was the last task for this issue, closing.

closed 2025-07-08 by BetsyMcPhail