RobotLocomotion/drake #22046: Support Vector3d scale for Mesh and ConvexMesh

RussTedrake, 2024-10-16:
This has [come up a number of times now](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1729022524445109). For me, it's specifically motivated by [#20444](https://github.com/RobotLocomotion/drake/issues/20444). In particular, I've seen a number of mujoco meshes which handle meshes for e.g. the fingers of a two-fingered gripper by using a single mesh and a reflection `scale=[-1,1,1]`.  

If the solution goes through in memory mesh files, that's fine for me. If it's better to just teach the proximity engine + renderers + visualizers how to use the `Vector3d scale`, that's fine for me, too. The immediate victory condition for me would be to
1) have the models in #20444 currently marked as "error 1" parse correctly, 
2) to have them simulate correct contact geometry,
3) to have them appear correctly in at least Meshcat, (e.g. with normals and textures)
4) to have them appear correctly in the default renderer.

If handling permutations of `{-1,1}` or permutations of `{-x,x}` is an easier partial resolution of the feature (perhaps it simplifies normals/texture coordinates?), and we throw on other cases, I would be ok with that. It would already cover my current need.

https://github.com/RobotLocomotion/drake/pull/22024 was a quick attempt, but it would have only helped for the 1 & 2, not 3 & 4.

chen-tianjian, 2024-11-27:
May I ask if there are any updates for this issue? We (Amazon) are interested in leveraging this feature.

RussTedrake, 2025-01-16:
Note that uniform scaling, which is negative, also isn't supported by MakeConvexHull.  I just found a model with `scale="-1 -1 -1"` which fails, despite being scalar valued.

RussTedrake, 2025-01-16:
> May I ask if there are any updates on this issue? We (Amazon) are interested in leveraging this feature.

@chen-tianjian -- I've got a work-around that is working very well for me now in 
https://manipulation.mit.edu/python/make_drake_compatible_model.html

See https://drake.mit.edu/troubleshooting.html for the 3 lines required to run it.

SeanCurtis-TRI, 2025-03-17:
Ok, I've got the kinks worked out; I've edited the description to include the PR strategy.

closed 2025-04-02 by SeanCurtis-TRI