RobotLocomotion/drake #22049: Get SignedDistancePair of Collision in SceneGraphCollisionChecker

cohnt, 2024-10-16:
The IRIS-NP2 algorithm (which we're hoping to implement in Drake) examines many configurations to check for collisions, and then uses those samples in collision to run the hyperplane search from IRIS-NP. However, knowing the configuration that's in collision is not enough -- we also need to know which pair of geometries in the world are in collision. If we want to use SceneGraphCollisionChecker (to leverage its speed and parallelism), then it would be extremely helpful if that object could return not just whether or not a configuration is in collision, but also the corresponding SignedDistancePair that was used to determine that there's a collision. [Note that this object is constructed as part of checking collisions](https://github.com/RobotLocomotion/drake/blob/a8b821c9649738d39f975ec0b1402782cf9f9610/planning/scene_graph_collision_checker.cc#L150) -- we just need a way to return it to the caller.

Related to #18830 and #21822. cc @rhjiang @wernerpe @AlexandreAmice @RussTedrake

AlexandreAmice, 2024-10-16:
[Linking the discussion from the Drake Developers Slack on this subject](https://drakedevelopers.slack.com/archives/C0557S260N8/p1719572390231669)

RussTedrake, 2024-10-17:
We also have a rough draft of how we were hoping to use it [here](https://github.com/rhjiang/drake/blob/95fd539b08fdc10e70926565edcdf1f9ec02397d/geometry/optimization/iris.cc#L543-L556); from [this branch](https://github.com/RobotLocomotion/drake/compare/master...rhjiang:drake:ray_iris).

cohnt, 2024-10-24:
Dropping back to leave a note for the future when I inevitably come back to this. The appropriate abstraction is to return a pair of bodies that are in collision, rather than a pair of geometries. Since this doesn't seem to be a major time cost in IRIS-NP2, I'm fine with closing the feature request as not planned.