RobotLocomotion/drake #22060: Need a tutorial on configuring lights/shadows for RenderEngineVtk

SeanCurtis-TRI, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
Configuring lighting in `RenderEngineVtk` can be tricky. It includes such interesting topics such as:

  - Environment maps
      - HDR vs LDR
      - Not only a source for reflection/refraction but introduce light energy into the scene
      - *Ambient* light.
  - Shadows
      - Not all lights cast shadows.
      - Not all lights cast shadows for all configurations.
      - Not all otherwise valid configurations will produce meaningful or even visible shadows.
      - There are practical workflows for configuring shadow-cast lights (and debugging them).
  - Balancing lights and environment maps
      - When using explicit lights and environment maps together, the amount of light energy must be "balanced".
  - Exposure
     - It is possible to introduce to much light energy into the scene and "over expose" the final image (lots of fully saturated pixels).
     - Tuning exposure will reduce and rebalance the energy in the final image (like changing aperture and shutter speed on a camera to reduce how much energy hits the film/ccd).

A tutorial to help end users understand how to better configure their lights would simultaneously better inform them of what is (and what is *not*) possible but also improve workflows so they can achieve a desired end result more efficieintly.

It can/should also include `CameraConfig` and `RenderEngineVtkParams`. Making mention of the other `RenderEngine` implementation is worthwhile but they need not be featured.

**Describe the solution you'd like**
Add to our deepnote tutorials.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
One of the single greatest challenges is being able to *show* the rendered images in the tutorial. (Or, akin to meshcat, be able to stream them to some other URL so that the user can open another browser window to see the image.) However, embedded would be preferred.

jwnimmer-tri, 2024-10-18:
> One of the single greatest challenges is being able to show the rendered images in the tutorial.

The [rendering_multibody_plant](https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/renderingmultibodyplant-642401c722c3411b8520755c3e5d79ae) tutorial already has examples of render engine images displayed inline in the notebook.  That's only still frames (not live video) but for a static tutorial that's the best we can do.

SeanCurtis-TRI, 2024-10-18:
Blech. I skimmed the tutorial list too quickly. I should've at least looked in the 'r's for "rendering".  Thanks.

closed 2025-04-16 by SeanCurtis-TRI

jwnimmer-tri, 2025-04-16:
BTW Remember to hand-work the "please run locally" edits into the Deepnote copy.  (I didn't see them yet.)