RobotLocomotion/drake #22070: Projected Gradient Descent Solver Interface

cohnt, 2024-10-23:
Projected gradient descent (PGD) is a strategy for solving constrained optimization problems -- one alternates between taking a step down the gradient of the objective function, and projecting onto the feasible set. In some contexts, this optimization strategy is preferable to using a second order solver like SNOPT or IPOPT. Having a native PGD solver in Drake would particularly be useful if we want to use it for the rounding stage of GcsTrajectoryOptimization.

I'm assigning to myself for now, but probably won't be able to get to this for some time. If someone wants it sooner, I'm happy to hand it off. Minimum features to include:

- User can specify solver used for the projection step
- User can manually specify a gradient function, instead of computing with autodiff/symbolic
- Backtracking line search

Additional features that might be nice:

- Specifying the affine hull in closed-form (can simplify the projection, or render it moot altogether)
- Momentum, and various other tricks that can lead to faster convergence

cc @shrutigarg914 @RussTedrake

cohnt, 2025-03-05:
cc @hongkai-dai if you have any feedback on how to do this, I'd appreciate it!

hongkai-dai, 2025-03-06:
I suppose one implementation would look like this
```cc
class PGDSolver {
 private:
  // Project x to the constraint set of prog
  VectorXd CalcProjection(const MathematicalProgram& prog, const Eigen::VectorXd& x) const;
  // Compute the descent direction of the cost of prog
  // If the user provides a direction (stored as a function pointer in `solver_options`, then use that direction. Otherwise, call autodiff to compute the direction.
  VectorXd CalcDescentDirection(const MathematicalProgram& prog, const Eigen::VectorXd& x, const SolverOptions& solver_options) const {
    if (solver_options.descent_direction_fun == nullptr) {
      AutoDiffVecXd x_ad = InitializeAutoDiff(x);
      AutoDiffVecXd y_ad;
      prog.cost().evaluate(x_ad, &y_ad);
      return ExtractGradient(y_ad);
    } else {
      return solver_options.descent_direction_fun(x);
    }
  }
  // Line search along the descent direction
  double LineSearch(const MathematicalProgram& prog, const VectorXd& x, const VectorXd& descent_direction) const;

  void Solve(const MathematicalProgram& prog, const VectorXd& x_init, const SolverOptions& solver_options, MathematicalProgramResult* result) const {
  x = x_init;
  while not_converged  {
    const VectorXd descent_direction = CalcDescentDirection(prog, x, solver_options);
    double eta = LineSearch(prog, x, descent_direction);
    x = CalcProjection(prog, x + eta * descent_direction);
  }
  }
};
```

Does this make sense to you?

I think one caveat of implementing PGDSolver as a sub-class of MathematicalProgramSolverInterface, is that MathematicalProgramSolverInterface is a const class after construction (its Solve() function is a const function). So you cannot store any information inside PGDSolver class. On the other hand, I often found that when I implement a solver, I want to store some information in the class for debugging/testing purpose, and also store some intermediate information that could be useful for the next iteration of the optimization.

So an alternative approach is to implement two classes, one `ProjectedGradientDescent` class that does all the mathematical computation and store the intermediate results, and another `PGDSolver` class that derives from `MathematicalProgramSolverInterface` class. `PGDSolver` class is a const class that doesn't store any information. It just calls `ProjectedGradientDescent`

```cc
class ProjectedGradientDescent {
  public:
    ProjectedGradientDescent(const MathematicalProgram& prog, std::function<...> descent_direction_fun);
    Eigen::VectorXd Projection(const VectorXd& x);
    double LineSearch(const VectorXd& x, const VectorXd& descent_direction);
    Eigen::VectorXd CalcDescentDirection(const VectorXd& x);
    Solve(const VectorXd& x, some_hyper_parameters, ..., some_return_values_like_primal_and_dual_solution);
  private:
    const MathematicalProgram* prog_;
};

const PGDSolver : public MathematicalProgramSolverInterface {
  public Solve(const MathematicalProgram& prog, const VectorXd& x_init, const SolverOptions& solver_options, MathematicalProgramResult* result) const {
    ProjectedGradientDescent(prog, ...);
    prog.Solver(x_init, ...,);
    // Fill in result 
  }
}
```
In this way, we can unit-test `ProjectedGradientDescent` public functions.

Also, one more question, we typically will compute the dual solution for the problem, how do you get the dual of the constraints through projected gradient descent?

cohnt, 2025-03-06:
What sort of information would you want to be storing between solves? I guess having some way of caching the projection program across solves could be useful, but we could handle that as an optional argument. I think we could implement something similar to VisualizationCallback if we wanted to debug intermediate iterations, but I don't have a great sense of what's that been like in your past experiences.

No idea about dual variables -- Shruti and I haven't needed them, so we never thought about them. Is that a requirement, or could we leave it as a TODO?

hongkai-dai, 2025-03-06:
> No idea about dual variables -- Shruti and I haven't needed them, so we never thought about them. Is that a requirement, or could we leave it as a TODO?

Yes, leaving it as a TODO is fine.

hongkai-dai, 2025-03-07:
> But all of this is within the scope of a single call to Solve, right? So that sort of information could be stored as a variable local to the method until the solve completes, and then discarded.

Yes, that is true, we do not have to create a new class for storing the intermediate information.

PR #22947 (2025-05-07): Implement a General Projected Gradient Descent Solver Interface
Closes #22070.

+@hongkai-dai for feature review. I doubt I got everything right on the first try, but hopefully this is a pretty good starting point.
cc @shrutigarg914 if you have any comments or suggestions as well.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22947)
closed 2025-05-07 by cohnt