RobotLocomotion/drake #22100: Operations on PiecewisePolynomials with different breaks

sadraddini, 2024-10-31:
Currently operations such as addition on piecewise polynomials are supported only if the objects have the same breaks. There have been discussions about the need for other operations such as stacking multiple piecewise polynomials into a new one. 

In order to support these operations, I am proposing some methods:
- [x] `AddBreak`: which adds "fake" breaks into a pp trajectory without modifying its values at anytime.
- [x] `Trim`: extracting a timed slice of a piecewise trajectory `trimmed_pp = pp[t_1,t_2]`. 

I have the features above in https://github.com/RobotLocomotion/drake/pull/22099

In future, I also plan to include:
- [ ] Stack (this is different from `StackedTrajectory` as it allows the user the maintain the object as piecewise polynomial).  
- [ ] Add support for `+=` when pp objects don't have the same breaks. 

Let me know your thoughts @RussTedrake

RussTedrake, 2024-10-31:
`AddBreak`, `Trim`, and `Stack` sound  very good to me. (thanks!)

The question I have is whether we should modify the existing `+=` to support objects that don't have the same break seems vs making a new method. Presumably the proposal is to call `AddBreak` repeatedly until the existing polynomial includes all of the breaks from `other`.  The functionality sounds very good, of course, I just don't know if it should be called out as a different method rather than the operator overload.

sadraddini, 2024-10-31:
I fully agree on the need for a different addition function for `+=`!