RobotLocomotion/drake #22107: Gurobi Always Printing to Console

cohnt, 2024-10-31:
### What happened?

I've recently observed that when Gurobi is used to solve a MathematicalProgram, the verbose output is always printed to the console. @wernerpe also observed this behavior. For methods that require solving many optimization problems, the output completely clogs the console, and can even crash a jupyter notebook.

### Version

16e7281bcd

### What operating system are you using?

Ubuntu 22.04, Other

### What installation option are you using?

compiled from source code using CMake, compiled from source code using Bazel

jwnimmer-tri, 2024-10-31:
One recent change was #21345, and perhaps I made a mistake there?  If you have time, you could try reverting that a let me know if it's to blame.

cohnt, 2024-10-31:
No console spam on 44bf559, yes console spam on 40adf9c4d7. So it sounds like that PR added the printing.

jwnimmer-tri, 2024-10-31:
Ah.  The old code set `LogToConsole` to 0 or 1 always.  The new code only ever sets it to `1`.  Maybe the Gurobi default is 1?

Does this solve it?

```patch
--- a/solvers/gurobi_solver.cc
+++ b/solvers/gurobi_solver.cc
@@ -954,9 +954,7 @@ void GurobiSolver::DoSolve2(const MathematicalProgram& prog,
 
   // Copy the remaining options into model_env.
   options->Respell([](const auto& common, auto* respelled) {
-    if (common.print_to_console) {
-      respelled->emplace("LogToConsole", 1);
-    }
+    respelled->emplace("LogToConsole", common.print_to_console ? 1 : 0);
     if (!common.print_file_name.empty()) {
       respelled->emplace("LogFile", common.print_file_name);
     }
```

jwnimmer-tri, 2024-10-31:
>  Maybe the Gurobi default is 1?

Yup.

https://www.gurobi.com/documentation/10.0/refman/logtoconsole.html#parameter:LogToConsole

I'll open a PR with that patch and ask for testing.

cohnt, 2024-11-04:
Now that I'm manually setting some additional Gurobi parameters (like number of threads), I'm getting spammed again -- this time with messages like `Set parameter Threads to value 1` and `Set parameter DualReductions to value 0`. Is it possible there's some order-dependent thing, where the console printing has to be disabled before other options?

jwnimmer-tri, 2024-11-04:
Sigh.  I imagine the full solution is to call a hard-coded `SetOptionOrThrow("LogToConsole", 0);` as early as possible during `DoSolve2`, before doing anything else.

jwnimmer-tri, 2024-11-04:
BTW I don't have a lot of time today, and I don't have convenient access to a Gurobi license to test, so I'd welcome any contributors' fixes.

cohnt, 2024-11-04:
I can confirm that dropping that in anytime before `options->CopyToCallbacks` is called in `gurobi_solver.cc` removes the extra print statements, but then those statements don't always come back if `kPrintToConsole` is set to true. (That doesn't bother me, but I can understand why it might bother someone.)

PR #22108 (2024-11-01): [solvers] Fix Gurobi console logging obey our defaults again
Closes #22107.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22108)
PR #22134 (2024-11-06): [solvers] Fix Gurobi console logging config to happen first
Closes #22107, again.

The bug was introduced subsequent to the most recent release, so we don't need a release note announcing the fix.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22134)
closed 2024-11-06 by sherm1