RobotLocomotion/drake #22120: CI coverage timeout //multibody/inverse_kinematics:differential_inverse_kinematics_integrator_test

DamrongGuoy, 2024-11-04:
### What happened?

Nightly CI has timeout after 600 seconds on a coverage test of //multibody/inverse_kinematics:differential_inverse_kinematics_integrator_test in:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-jammy-gcc-bazel-nightly-coverage/398/

On my Jammy computer, the test passed in 400 seconds:
```
time bazel test --config=gcc --compilation_mode=dbg --config=kcov  //multibody/inverse_kinematics:differential_inverse_kinematics_integrator_test
```

### Version

d6ff6cc1fa1e5c7077e82f5c8de605e23adc7ea4

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
TIMEOUT

Using drake_cc_googletest_main.cc

[==========] Running 7 tests from 2 test suites.
[----------] Global test environment set-up.
[----------] 6 tests from DifferentialInverseKinematicsIntegratorTest
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.BasicTest
[2024-11-04 12:53:50.650] [console] [warning] Differential IK started returning status:"stuck" at time 0.
[       OK ] DifferentialInverseKinematicsIntegratorTest.BasicTest (1031 ms)
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.FrameATest
[       OK ] DifferentialInverseKinematicsIntegratorTest.FrameATest (355 ms)
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.ParametersTest
[       OK ] DifferentialInverseKinematicsIntegratorTest.ParametersTest (224 ms)
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.DifferenceEquationSystemTest
[       OK ] DifferentialInverseKinematicsIntegratorTest.DifferenceEquationSystemTest (234 ms)
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.InitializationEvent
[       OK ] DifferentialInverseKinematicsIntegratorTest.InitializationEvent (276 ms)
[ RUN      ] DifferentialInverseKinematicsIntegratorTest.UseRobotStatePort
[2024-11-04 12:53:51.983] [console] [warning] Differential IK started returning status:"stuck" at time 0.
[2024-11-04 12:53:51.987] [console] [warning] Differential IK started finding solutions again at time 0.
[       OK ] DifferentialInverseKinematicsIntegratorTest.UseRobotStatePort (239 ms)
[----------] 6 tests from DifferentialInverseKinematicsIntegratorTest (2364 ms total)

[----------] 1 test from DeprecatedTest
[ RUN      ] DeprecatedTest.BasicTest
[2024-11-04 12:53:52.217] [console] [warning] InputPort[1] (X_WE_desired) of System ::_ (DifferentialInverseKinematicsIntegrator) is deprecated: Use the `X_AE_desired` input port instead.
[2024-11-04 12:53:52.222] [console] [warning] Differential IK started returning status:"stuck" at time 0.
[       OK ] DeprecatedTest.BasicTest (233 ms)
[----------] 1 test from DeprecatedTest (233 ms total)

[----------] Global test environment tear-down
[==========] 7 tests from 2 test suites ran. (2603 ms total)
[  PASSED  ] 7 tests.
kcov: 119 invalid breakpoints skipped in /media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/_bazel_ubuntu/13ab93db129735c0cc7824860397095a/execroot/drake/bazel-out/k8-dbg/bin/multibody/inverse_kinematics/differential_inverse_kinematics_integrator_test
kcov: 1 invalid breakpoints skipped in /lib/x86_64-linux-gnu/libm.so.6
kcov: 1 invalid breakpoints skipped in /lib/x86_64-linux-gnu/libc.so.6
kcov: 1 invalid breakpoints skipped in /lib64/ld-linux-x86-64.so.2

//multibody/inverse_kinematics:differential_inverse_kinematics_integrator_test TIMEOUT in 600.0s

  /media/ephemeral0/ubuntu/workspace/linux-jammy-gcc-bazel-nightly-coverage/_bazel_ubuntu/13ab93db129735c0cc7824860397095a/execroot/drake/bazel-out/k8-dbg/testlogs/multibody/inverse_kinematics/differential_inverse_kinematics_integrator_test/test.log
```

DamrongGuoy, 2024-11-04:
Close the issue since it happened the first time.

DamrongGuoy, 2024-11-12:
Reopen since it repeated on November 6, 2024:
https://drake-jenkins.csail.mit.edu/job/linux-jammy-gcc-bazel-nightly-coverage/400/

CC: @nicolecheetham

rpoyner-tri, 2026-11-10:
last complaint was against Jammy, which is now gone. Closing.

closed 2026-11-10 by rpoyner-tri