RobotLocomotion/drake #22131: Update Coefficients in Cost with Copying Whole Vector

cohnt, 2024-11-05:
In scenarios where we're solving a sequence of similar mathematical programs (e.g. #22084), we have to copy over the entire coefficient vector to update the cost. It would be much more efficient if we could just modify the internally held coefficient vector in-place. I specifically care about LinearCost, but I imagine it would be applicable to other costs as well.

cc @AlexandreAmice

hongkai-dai, 2024-11-11:
@cohnt  Currently `UpdateCoefficients` just copies the new coefficient vector to the internally held vector. I think you are suggesting something like this
```cc
LinearCost::UpdateCoefficientEntry(int i, double val) {
  a_[i] = val;
}
```
I wonder why this "update-in-place" is more efficient than copying a vector?

cohnt, 2024-11-11:
That method would be perfect for what I need.

If we have a large vector and we only need to change one or two entries of it (as in the generic `ConvexSet::IsBounded` method), I would think it's significantly faster to just modify the entry, rather than keeping an entirely separate vector around, changing the entry there, and then copying it in.

cohnt, 2024-11-20:
FYI I only needed this feature for LinearCost, so I'm willing to close it now that #22152 has merged. But @hongkai-dai wants to keep this open until all costs that can support it do support it, that's okay with me.

PR #22260 (2024-12-11): update individual entry in A/b of some costs.
Including LInfNormCost, L1NormCost, PerspectiveQuadraticCost.

Resolves #22131

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22260)
closed 2024-12-11 by SeanCurtis-TRI