RobotLocomotion/drake #22176: xcode16 vs. RandomSource ==> linker errors

rpoyner-tri, 2024-11-14:
### What happened?

```
$ bazel build  //systems/primitives:random_source_test
...
Undefined symbols for architecture arm64:
  "drake::systems::RandomSource<drake::symbolic::Expression>::~RandomSource()", referenced from:
      vtable for drake::systems::RandomSource<drake::symbolic::Expression> in random_source_test.o
  "drake::systems::RandomSource<drake::symbolic::Expression>::~RandomSource()", referenced from:
      vtable for drake::systems::RandomSource<drake::symbolic::Expression> in random_source_test.o
  "drake::systems::RandomSource<drake::symbolic::Expression>::SetRandomState(drake::systems::Context<drake::symbolic::Expression> const&, drake::systems::State<drake::symbolic::Expression>*, drake::RandomGenera\
tor*) const", referenced from:
      vtable for drake::systems::RandomSource<drake::symbolic::Expression> in random_source_test.o
  "drake::systems::RandomSource<drake::symbolic::Expression>::SetDefaultState(drake::systems::Context<drake::symbolic::Expression> const&, drake::systems::State<drake::symbolic::Expression>*) const", referenced\
 from:
      vtable for drake::systems::RandomSource<drake::symbolic::Expression> in random_source_test.o
```

It definitely has something to do with `is_symbolic_convertible` in the unit test. I've tried to peel things apart further, but I feel like I'm getting contradictory results as I delve down into the implementations of `is_dynamic_castable` and `NiceTypeName`.

For a full CI build with the same failure, see  https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-unprovisioned-clang-bazel-experimental-release/9/consoleFull

### Version

master circa 1.34

### What operating system are you using?

macOS 14 (Sonoma)

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2024-11-14:
My guess is that instead of `EXPECT_FALSE(is_symbolic_convertible ...)` we should do `EXPECT_TRUE(is_not_symbolic_convertible .. )` -- by adding a new helper function ("is_not_...") for that purpose.

rpoyner-tri, 2024-11-14:
What I haven't yet investigated are the other existing instances of `EXPECT_FALSE(is_symbolic_convertible)`, why they don't fail, and why they might be different.

jwnimmer-tri, 2024-11-14:
That's a fair question, but `dynamic_cast<MyNonSymbolicSystem<Expression>>` is unquestionably invalid code, so I was just thinking we stop doing that, see what happens, and maybe just move on if it works.

rpoyner-tri, 2024-11-14:
So, the difference that matters is ... RandomSource is declared `final`. The other potential victims are not.

It appears that (for apple clang 16 alone afaict) `final` means "fill out my instantiation's vtable, no matter what". Ubuntu (all compilers tried) proceed just fine.

rpoyner-tri, 2024-11-16:
Fixed with merge of #22188.

closed 2024-11-16 by rpoyner-tri