RobotLocomotion/drake #22192: issue with meshcat

Dawitsco, 2024-11-15:
### What happened?

I have issues with the meshcat local serve not opening in my browser(Chrome). I followed the drake tutorial on "Authoring a Multibody Simulation" I noticed when  the visualizer is set to the class ModelVisualizer(meshcat=meshcat) and visualizer.Run() is called the meshcat opens but when i use this approach                                                                                                                                          
AddDefaultVisualization(builder,meshcat)
diagram =builder.Build()
simulate = Simulator(diagram)
simulate.Initialize()
meshcat.StartRecording()
simulate.AdvanceTo(2)
meshcat.PublishRecording()
it provides the meshcat listening link but when i try to open in my browser it says "the site can't be reached". any solution to this. I am using ubuntu and using VScode IDE

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

jwnimmer-tri, 2024-11-15:
For reasons I don't fully understand, somehow using Meshcat and VSCode together can sometimes be troublesome.

For starters, please try to reproduce the problem without using VSCode.

If you can still reproduce the problem, then please post instructions here for how we (the Drake developers) can reproduce the problem for ourselves.  The only way we can fix bugs is when the bug report contains enough detail for us to see the the problem for ourselves.

jwnimmer-tri, 2026-09-10:
Closing for lack of details.

closed 2026-09-10 by jwnimmer-tri