RobotLocomotion/drake #22196: Curvilinear Joints

mshalm, 2024-11-15:
**Is your feature request related to a problem? Please describe.**
I would like to simulate robots moving via a motorized carriage on a curved rail, e.g. the Hebco 7th axis system shown below.

[image]

**Describe the solution you'd like**
To us, the most natural description of the motion of these mechanisms is as a single-DoF Joint<T> parameterized by the curve (following [this paper](https://nmbl.stanford.edu/publications/pdf/Seth2010.pdf)), with state capturing distance traveled and tangential velocity along the curve. This allows for existing Joint-based infrastructure like Implicit PD feedback; damping; and reflected inertia to be utilized to simulate the behavior.

All other Joint<T>’s are implemented to have SI-unit states; following this pattern requires a path-distance-parameterized curve. Generalized forces would then be equivalent to a force pair applied at the rail, tangent to the curve. Our prototype implementation of this joint and suggested strategy for drake is: 
1. To implement a Trajectory<T> which models the curve. We find that piecewise circular arcs / line segments within a plane to be one case where path length parameterization can be analytically calculated efficiently (shown below). This parameterization is also quite general, in that many curves can be approximated with several circular/linear segments.
2. To implement a corresponding Joint<T> and Mobilizer<T> which are thin wrappers of the trajectory.
3. To support URDF/SDF parsing of the joint via the radii $r_i$ (equivalently, curvature $\kappa_i = \frac{1}{r_i}$) /lengths $l_i$ of the segments.

Piecewise linear/circular rail birds-eye view:

[image]

Isometric view:

[image]

We intend to submit 3 PR’s mapping to these 3 components for our prototype, unless Drake developers feel strongly that a different implementation is required.

**Describe alternatives you've considered**
Another alternative would be to mount the robot to a floating base, and use feedback control to mimic the behavior of the real joint. We expect this to require a very stiff controller, and it adds 12 DoF to the state rather than the 2 required for the direct implementation of the joint. As such it is unlikely to be performant.

**Additional context**
Reiterating that we intend to send in the PR’s ourselves if Drake developers are happy to receive an implementation of this strategy.

sherm1, 2024-11-15:
Beautiful!
cc @mitiguy @amcastro-tri

rpoyner-tri, 2025-01-31:
Per discussion in #22469, let's keep this issue open to track completion of test coverage (diagnostics branches) and documentation of new `drake:` tags.

I'm stealing assignee role from @sherm1 to track these.

PR #22840 (2025-04-01): [parsing] Fix parsing of erroneous curvilinear joint input
Closes: #22196.

Merge is blocked until #22839 is merged and this patch is rebased.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22840)
closed 2025-04-01 by jwnimmer-tri