RobotLocomotion/drake #22233: IrisInConfigurationSpaceFromCliqueCover crashes when ellipsoid center is in collision

dkreho1, 2024-11-23:
### What happened?

IrisInConfigurationSpaceFromCliqueCover crashes with the following exception:
`what():  Failure at geometry/optimization/hyperellipsoid.cc:403 in CheckInvariants(): condition 'this->ambient_dimension() == A_.cols()' failed.`

I've looked into the source code, and it appears that this issue occurs when the ellipsoid center is in collision. In this case, the code attempts to construct a new Hyperellipsoid centered at one of the clique nodes:
https://github.com/RobotLocomotion/drake/blob/1c5e3415c9a30c0bf15c4cc13d2173c0019dc865/planning/iris/iris_from_clique_cover.cc#L232

However, it seems the constructor parameters are passed in the wrong order:
https://github.com/RobotLocomotion/drake/blob/1c5e3415c9a30c0bf15c4cc13d2173c0019dc865/geometry/optimization/hyperellipsoid.cc#L40-L44

I haven't tested this directly, as I'm running the pre-compiled binaries, but it seems like this is the only thing that could be causing the issue.

### Version

1.34.0

### What operating system are you using?

Ubuntu 22.04

wernerpe, 2024-11-25:
I have a reproduction of the error locally, and am working on a fix [here](https://github.com/wernerpe/drake/tree/bugfix/vcc ) .

wernerpe, 2024-11-26:
#22239 Takes care of this bug. I am closing the issue. Thanks again :+1:

closed 2024-11-26 by wernerpe