RobotLocomotion/drake #22251: [ci] Save undeclared test outputs

jwnimmer-tri, 2024-12-02:
**Is your feature request related to a problem? Please describe.**

Sometimes, tests fail in CI but the failure is not reproducible locally.  In this situation, the only diagnostic tool we have to solve the problem is the test output logged in CI.  In many cases all we need is the console output (text), but in some cases we also need to inspect binary data (e.g., images, compressed logs, etc.)

At the moment, there is no way to retrieve non-text test output when Jenkins runs our tests and uploads the test results into CDash, but sometimes we need that information in order to solve the problem.

**Describe the solution you'd like**

In Bazel, extra test output files (beyond the normal assertion printouts) are called "undeclared outputs" as documented at https://bazel.build/reference/test-encyclopedia.

For example, after running the `internal_render_engine_vtk_test` there is a file named `drake/bazel-testlogs/geometry/render_vtk/internal_render_engine_vtk_test/test.outputs/outputs.zip` and inside that zipfile there are images like `CloneIndependence/line_1318_Clone_independence_label.png` which have been saved by the test program.

The request here is to allow a Drake Developer to download that file (`outputs.zip`) when a test fails in CI.

Note that we don't need to keep these files for passing tests, only failing ones.

If it's easy, then it would also be nice for the upload to be associated in CDash so that CDash users can directly refer to it.  If that's difficult, then uploading it only to Jenkins is perfectly fine.

**Describe alternatives you've considered**

Have the test programs print the binary files as base64-encoded text, which the developer would copy-paste and then decode.

**Additional context**

For reference, here is TRI's groovy code from our `Jenkinsfile` that uploads the zipfiles when CI fails.  With this snippet, the user can download the zipfile from the Jenkins PR result page.

```groovy
pipeline {
  stages {
    stage('Parallel') {
      parallel {
        stage('...') {
          agent { ... }
          stages {
            ...
            stage('Test') {
              ...
            }
          }
          post {
            failure {
              junit 'bazel-testlogs/**/*.xml'
              archiveArtifacts artifacts: 'bazel-testlogs/**/test.outputs/outputs.zip', allowEmptyArchive: true
            }
          }
        }
      }
    }
  }
}
```

jwnimmer-tri, 2024-12-10:
Two important caveats:

- With bazel 8 (about to land now), tne `test.outputs` is no longer zipped by default.  We can turn on a flag to zip it again, but probably best if we can deal with the unzipped outputs directly.

- This feature must NOT be enabled for coverage (kcov) builds.  Coverage builds have a ton of extra output files, which we do not want to keep around (very large).  We already upload the coverage summary, which is enough.  So we'll need some kine of on/off flag for this.

jwnimmer-tri, 2025-03-03:
From f2f -- doing this the "right" way of transmogrifying bazel json to ctest xml has gaps and is somewhat difficult, so for now we'll just upload the files associated with the "build" instead of each test one by one.

BetsyMcPhail, 2025-03-24:
The initial implementation in #311 revealed a defect in CDash that caused the queue to slow down and get backed up. A hot fix has been implemented to resolve things for the moment. Reopening this issue while the longer-term solution is being worked on.

williamjallen, 2025-03-25:
I just opened https://github.com/Kitware/CDash/pull/2749 with the long-term fix for this issue.

BetsyMcPhail, 2025-03-26:
Closing this issue as the short-term fix seems to be successful. We will create a new issue once CDash 3.10 is release to apply the long-term fix.

closed 2025-03-26 by BetsyMcPhail