RobotLocomotion/drake #22308: Wrong date in macOS packaging nightlies

jwnimmer-tri, 2024-12-13:
In the [the log](https://drake-jenkins.csail.mit.edu/view/Packaging/job/mac-arm-sonoma-unprovisioned-clang-cmake-nightly-packaging/73/consoleFull) from last night's nightly packaging build, we can see this:

```
[10:31:08 PM]  -- Upload complete: https://drake-packages.csail.mit.edu/drake/nightly/drake-0.0.20241212-mac-arm64.tar.gz
```

It should have said `20241213` not `20241212` (off by one day).

I assume this happens because the timezone in the new images is set incorrectly.

williamjallen, 2024-12-13:
I just confirmed that the timezone is set to `US/Pacific` in one of the currently running macOS VMs.  I'll investigate further.

williamjallen, 2024-12-13:
I fixed the timezone in the Xcode base image, rebuilt the entire stack on top of it, and activated the new images in CI.

closed 2024-12-13 by williamjallen