RobotLocomotion/drake #22312: Direct collocation not correctly converting scene graph configuration values

marip8, 2024-12-14:
### What happened?

After updating from Drake 1.35.0 to 1.36.0, I encounter a run-time error when using the `DirectCollocation` optimization that says:

```
RuntimeError: Exception while evaluating SNOPT costs and constraints: 'GetPropertyOrDefault(): The property ('material', 'hunt_crossley_dissipation') exists, but is of a different type. Requested 'drake::AutoDiffXd', but found 'double''
```

and

```
RuntimeError: Exception while evaluating SNOPT costs and constraints: 'GetPropertyOrDefault(): The property ('material', 'point_contact_stiffness') exists, but is of a different type. Requested 'drake::AutoDiffXd', but found 'double''
```

It appears that this might be caused to the changes in default values in the scene graph configuration added by [this PR](https://github.com/RobotLocomotion/drake/pull/21463). If I reset the values to `hunt_crossley_dissipation=None` and `point_contact_stiffness=None` (their values in Drake 1.35.0), then my `DirectCollocation` optimization runs correctly again.

Is the issue that the `DirectCollocation` class is not properly casting these scene graph configuration values from `double` to `AutoDiffXd`, or is there just a fundamental conceptual problem using these two values in the optimization?

### Version

1.36.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

jwnimmer-tri, 2024-12-14:
I think the problem is `MultibodyPlant` accessing the values using the wrong type annotation.  The properties should be accessed using `double` (not `T`) or could be omitted and `double` would be correctly inferred:

https://github.com/RobotLocomotion/drake/blob/d819ff20933392a1a81b9b9ab0d218d078b20c63/multibody/plant/multibody_plant.cc#L1568-L1575

I think you found the best workaround for now, by setting `hunt_crossley_dissipation=None` and `point_contact_stiffness=None` in the config.

jwnimmer-tri, 2024-12-14:
Hmm, I just found #13630 which led me to the docs at https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#accessing_contact_properties, which specifies that the properties are of type `T` not `double`.  So maybe the code which sets defaults in SceneGraph is actually what's wrong.