RobotLocomotion/drake #22322: something is blocking LCM traffic on macOS Sequoia CI

rpoyner-tri, 2024-12-17:
### What happened?

https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-bazel-continuous-release/1/consoleFull for example.

Victory condition is to determine what config is needed for LCM multicast to work, and to apply it to CI images and/or setup scripts.

### Version

v1.36.0

### What operating system are you using?

macOS 15 (Sequoia)

### What installation option are you using?

compiled from source code using Bazel

rpoyner-tri, 2024-12-17:
I'll do the initial investigation in a few days, then maybe some of the fix application will get delegated to kitware folks.

ggould256, 2024-12-17:
The extent of my research:
 * https://developer.apple.com/documentation/technotes/tn3179-understanding-local-network-privacy#macOS-considerations
   * Afaict the "launchd agent" text there applies to sshd, and I think also to jenkins workers.
   * However it also says "The multicast entitlement isn’t required on macOS."
   * But maybe that's out of date.  If so, the relevant iOS instructions probably apply; https://developer.apple.com/forums/thread/663271

If these hypotheses are correct, it is likely that drake works when run from the terminal but not when run by Jenkins or over ssh, and that the remedy probably involves our code signing workflow, or changing something about the relevant user.

jwnimmer-tri, 2025-01-13:
@rpoyner-tri maybe it's time to delegate to kitware now?

tyler-yankee, 2025-01-28:
It seems that it's a network configuration issue for the CI images.

* Failed to reproduce when running locally on macOS Sequoia
* Failed to reproduce when running over ssh on macOS Sequoia
* Reproduced when running on the macOS Sequoia images

I applied the fix pointed out here: https://github.com/lcm-proj/lcm/issues/476. 
(@jwnimmer-tri actually chimed in on a [related issue](https://github.com/lcm-proj/lcm/issues/60) which led me here.)
i.e., explicitly enabling multicast traffic on loopback interface as follows:
```
sudo route -nv delete 224.0.0.0/4
sudo route -nv add -net 224.0.0.0/4 -interface lo0
```

After changing the network configuration the test passes, and applying this change to the macOS doesn't break anything to run locally or over ssh either. So, we should update all the macOS Sequoia CI images with this network configuration, verify tests still passing on the images and locally, and then re-enable that unit test.

jwnimmer-tri, 2025-01-28:
> ... update all the macOS Sequoia CI images with this network configuration ...

Perhaps goes without saying, but we should also ensure that it _stays_ updated, which means either adding to the macOS CI image setup scripts, or at least the macOS CI image setup documentation.

Ideally we would also update our user-facing documentation to warn macOS users about this problem (maybe link to that issue), but I don't immediately know where we could/should put that.

tyler-yankee, 2025-01-28:
Yeah, we can describe this in the macOS CI image setup documentation once completed. I don't believe the setup scripts need to be changed at this point. The troubleshooting page you linked above might be a good place to link the issue and provide a brief description?

tyler-yankee, 2025-01-28:
Waiting to update the macOS CI images until the updates from #22477 (see: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-unprovisioned-clang-bazel-nightly-release/44/) are ready. cc: @williamjallen

closed 2025-02-10 by jwnimmer-tri