RobotLocomotion/drake #22326: 3rd-party 3D Constrained Delaunay Triangulation/Tetrahedrization (CDT)

DamrongGuoy, 2024-12-18:
Placeholder to study how to include 3D CDT library into Drake.

DamrongGuoy, 2025-01-13:
https://github.com/MarcoAttene/CDT is used in a 2023 SIGGRAPH Asia paper. According to that paper, it is about 8.6% more robust than TetGen (https://github.com/libigl/tetgen).

Like TetGen, this code has GPL/LGPL licenses.

DamrongGuoy, 2025-01-13:
https://github.com/FabienPean/VegaFEM/blob/main/libraries/mesher/tetMesher.h is a re-implementation of TetGen with compatible licenses with Drake.

Nobody updated Vega FEM (https://viterbi-web.usc.edu/~jbarbic/vega/) at the University of Southern California for nearly five years.  The latest release was v4.0 in 2018 (https://viterbi-web.usc.edu/~jbarbic/vega/download.html).

https://github.com/FabienPean/VegaFEM is an unofficial fork of Vega FEM 4.0.

jwnimmer-tri, 2025-03-17:
@DamrongGuoy @amcastro-tri is this still "in progress"?  I get the impression from recent discussions that the focus has moved elsewhere, but this is still listed as the top priority on our board.