RobotLocomotion/drake #22386: Build 'Convex' shape in memory from vertices

Akshay5312, 2025-01-03:
**Is your feature request related to a problem? Please describe.**
There is no straightforward way to create a general 'Convex' shape given vertices.This means that to visualize shapes generated during runtime, one needs to generate a mesh file in memory (```MemoryFile(formatted_string_containing_vertices, ".OBJ", filename_hint)```) then parse that file into a Convex object. This is inconvenient.

**Describe the solution you'd like**
As suggested by @jwnimmer-tri, a ```Convex::Convex(const Matrix3d& vertices,  const std::string& filename_hint, double scale = 1.0)``` constructor would be great. This would also help implement a method to convert VPolytopes to Convex shapes to register VPolytopes as geometries.

jwnimmer-tri, 2025-01-03:
I imagine the best signature will be:

```c++
explicit Convex(const Matrix3X<double>& vertices, std::string filename_hint = {}, const Vector3d& scale = Vector3d::Ones());
```

I think it's okay to default the hint to empty -- I don't think most users will need pass anything in since in most cases the `Convex` will be wrapped into a `GeometryInstance` with its own name.

For the scale we're [moving towards non-isotropic scaling](https://github.com/RobotLocomotion/drake/issues/22046) so I suggest `Vector3d` instead of `double`.

Akshay5312, 2025-01-03:
> For the scale we're [moving towards non-isotropic scaling](https://github.com/RobotLocomotion/drake/issues/22046) so I suggest `Vector3d` instead of `double`.

Currently, the Convex shape has a scalar 'scale' member (it is accessed and used otherwise as well). Changing this to a Vector3d would likely break a few things.

Would you recommend I implement non-isotropic scaling by directly scaling the vertices before parsing them into a string?

jwnimmer-tri, 2025-01-03:
Right.  I imagined that for now, we would multiply the scale with vertices before making the wavefront string, and store `scale_ = 1.0` internally.  Then later once the member field is changed we could undo that.  But maybe that's over-complicating things and we should just stick with `scale double = 1.0` for now, and we can overload with a Vector3d constructor in the future.

jwnimmer-tri, 2025-01-14:
Fixed by #22390.

closed 2025-01-14 by jwnimmer-tri