RobotLocomotion/drake #22403: Pip install on Amazon Linux 2023

jwnimmer-tri, 2025-01-06:
**Is your feature request related to a problem? Please describe.**

Sometimes, users want to run `pydrake` inside an AWS Lambda service.  Those lambda containers are based on Amazon Linux.  See [What is Amazon Linux 2023?](https://docs.aws.amazon.com/linux/al2023/ug/what-is-amazon-linux.html) for details.

Currently, this awkward / impossible because our wheels require `libc >= 2.35` but AL2023 only provides `libc == 2.34`.

**Describe the solution you'd like**

Update our wheel build so that our `manylinux` wheels can run on `public.ecr.aws/lambda/python:3.12`.

Running the wheel build in a docker container with a lower libc6 baseline (so that the produced will has a lower baseline) will be relatively easy.  The question is whether other build-dependencies like GCC or Clang or libstdc++ or etc. are sufficient, and how difficult it would be to support AL2023 wheels long term.  That will require some investigation.

**Describe alternatives you've considered**

None.

**Additional context**

See AL2023 docs [Improved toolchain: gcc, binutils, and glibc](https://docs.aws.amazon.com/linux/al2023/ug/glibc-gcc-and-binutils.html) for a version baseline.

jwnimmer-tri, 2025-01-20:
On paper at least, the `manylinux_2_34` base images from https://github.com/pypa/manylinux seem like they should be good enough to build Drake (after a few more removals of dead code and OS dependencies already in flight).

mwoehlke-kitware, 2025-03-11:
Curiously, `manylinux_2_34` has (by default) Python 3.9, and also 3.11 and 3.12, but not 3.10. So I guess we need to build it from source? (What version? 3.10.0, Jammy's 3.10.12, or something else?) Should we use the provided 3.12?

Maybe we should just always build Python?

jwnimmer-tri, 2025-03-11:
> (What version? 3.10.0, Jammy's 3.10.12, or something else?

I imagine it doesn't matter at all, so dealer's choice.  If I had to pick, I'd use the freshest patch release.

> Should we use the provided 3.12?

Sure.  Probably the less customization we do, the smother things will be?

mwoehlke-kitware, 2025-03-12:
> I assume it's easier to use what they already have.

We already build 3.12. The *development* effort to continue doing so is zero. It's just a matter of specifying the patch version and SHA to build 3.10 and 3.11. From a *development* standpoint, I think there's an argument to quit faffing about with what the distro does or doesn't provide and just always build Python.

It also justifies always installing the Python build dependencies, which I'd like to do because I already have two code paths that have sprouted logic to deal with package manager selection and didn't want to have to add that to `build-python.sh` also. (Although, do we even want to keep `apt` support? Or just replace it with `dnf`?)

p.s. I feel like I remember an in-flight PR that makes changes to `build-python.sh`?

jwnimmer-tri, 2025-03-12:
> ... always build Python.

The Python binary has two touchpoints: Drake <=> Python and Python <=> OS image.  The Drake <=> Python interface has nearly no constraints / coupling; we can take any patch version or build options and be happy.  My inclination to prefer the OS image was for the second touchpoint -- so that we never need to worry about fixing / configuring / tuning our build script for Python to align with the OS image.  If upstream already did that for us, and will continue to maintain it across other OS changes, that's where I imagined the cost savings.

On the other hand, maybe neither touchpoint will care, so yes -- I'm okay with always building all versions from source for now.  If we discover that it's troublesome to bootstrap or maintain that design, then we can switch back to the OS images later.

Question, though: do the container images tweak the installed Python (the site-packages config, pip version, setuptools something, etc.) to be optimized for making packages?  That would be a good reason to prefer the OS install of the Python _ecosystem_ over an ecosystem that we roll by hand.  In short, these images are _supposed_ to be the pre-baked easy way to just run a build and be happy, so any of that entropy which they've already solved for us, we should try to use if we can.

> (Although, do we even want to keep apt support? Or just replace it with dnf?)

For the container building the wheels, it's OK to completely repave how we prep things, which I imagine means not using apt.

For the container we _test_ the wheel on at the end, that should still continue to test on Ubuntu (+ apt).

> p.s. I feel like I remember an in-flight PR that makes changes to build-python.sh?

The https://github.com/RobotLocomotion/drake/pull/22580 hasn't landed yet.

mwoehlke-kitware, 2025-03-12:
> Do the container images tweak the installed Python to be optimized for making packages?

AFAICT they're just stock AlmaLinux packages.

> If we discover that it's troublesome to bootstrap or maintain that design, then we can switch back to the OS images later.

:+1:

mwoehlke-kitware, 2025-03-12:
I think the next step here might be getting regular Drake builds working on not-Ubuntu. The first error I'm hitting is that Drake still depends on `lsb_release`.

mwoehlke-kitware, 2025-03-12:
Gross. Manylinux *does* add a whole bunch of Pythons on top of the AlmaLinux base. Which... don't ship libraries, and are thus useless for Drake.

PR #22779 (2025-03-25): Build wheels on AlmaLinux instead of Ubuntu
Change wheel builds to use AlmaLinux instead of Ubuntu as the host platform. The motivating effect of this is to "downgrades" our required glibc version from 2.35 to 2.34.

Note that AlmaLinux 9 is the base of manylinux_2_34_x86_64:2025.01.05-1. The reason we are not using manylinux is because a) it provides Python interpreters that do not have headers or libraries, and so are not useful for building Drake, and b) its compilers interfere with use of the underlying AlmaLinux compilers and attempts to use them result in obscure failures.

Fixes #22403. [... cut]
closed 2025-03-25 by rpoyner-tri