RobotLocomotion/drake #22415: debug-only assertions in rotational inertia are hostile to uses in parsing

rpoyner-tri, 2025-01-07:
### What happened?

Debug-only error checking behavior in RotationalInertia:
* forces me to bifurcate tests on compilation mode
* is generally hostile to the rule that errors in parsing flow to a diagnostic policy, rather than just throwing.

TBD: example of a split test, investigation into execution paths and alternatives.
Victory condition: parser unit tests don't need to use `#ifdef` to capture divergent behavior based on compilation mode.

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2025-01-07:
Yup, I've been pondering this, too.

The purpose of `DRAKE_ASSERT` is to identify bugs inside Drake (in our Debug CI jobs).  Using `DRAKE_ASSERT` to sanitize data from users is wholly inappropriate.

rpoyner-tri, 2025-01-07:
Example of a split test case is in #22414 .

rpoyner-tri, 2025-01-08:
The problem seen in early revisions of #22414 was able to be avoided, but the potential for surprises remains. I'll need to study this a bit more to better reframe the problem.