RobotLocomotion/drake #22440: Locate normal contact force from hydroelastic pressure field to ensure equivalent force/torque.

mitiguy, 2025-01-10:
For a single contact polygon, (with no friction and no penetration rate), it seems that the location of the single force (which replaces the force/torque on the contact polygon) is currently (in Drake) located at the centroid of the contact polygon.   Although this seems reasonable for a uniform pressure distribution on the contact polygon, it is questionable if this is valid when the pressure distribution is **not** uniform, particularly when the pressure distribution varies greatly from one vertex to another (e.g., 0 pressure to very high pressure). 

For example, consider a contact polygon which is a single rectangle whose vertices have a pressure distribution of 0 N/m²,  0 N/m²,  2.0E7 N/m²,  2.0E7 N/m².  Replacing this pressure distribution with a single force at the rectangle's centroid is questionable (it seems the correction location is 2/3 of the way towards the vertices having 2.0E7 N/m² pressure).

sherm1, 2025-01-10:
This suggests that we may be calculating contact moments incorrectly when there are just a few large polygons in a hydroelastic contact surface. @amcastro-tri, did we discuss this when we decided to use first-order quadrature? I don't remember it coming up. With a linear pressure field, we are getting the total force on the polygon right, but then we calculate moments assuming that that force is applied at the centroid which seems wrong given Paul's comment above.

amcastro-tri, 2025-02-12:
FYI.
I developed the equations in dyadic form and implemented in this [branch](https://github.com/amcastro-tri/drake/tree/unstable_cube) as a possible alternative to fix #22213. 
While it does not fix that issue (details in that specific issue), I figured you'd like to have this implementation as a baseline.

I'll update here with the equations for reference later today or early tomorrow.