RobotLocomotion/drake #22444: ModelVisualizer throws when a model has implicit PD gains

RussTedrake, 2025-01-12:
### What happened?

It should be easy to construct a more minimal example, but in the spirit of not losing my in-hand reproduction, I ran:
```
from pydrake.all import ModelVisualizer, PackageMap, StartMeshcat, DiscreteContactApproximation
from manipulation.make_drake_compatible_model import MakeDrakeCompatibleModel

visualizer = ModelVisualizer(meshcat=meshcat)
plant = visualizer.parser().plant()
plant.set_discrete_contact_approximation(DiscreteContactApproximation.kLagged)
package_map = visualizer.parser().package_map()
package_map.AddRemote(
    package_name="mujoco_menagerie",
    params=PackageMap.RemoteParams(
        # This repository doesn't have up-to-date tags/releases; the scary
        # hash in the url is the most recent commit sha at the time of my
        # writing.
        urls=[
            f"https://github.com/google-deepmind/mujoco_menagerie/archive/469893211c41d5da9c314f5ab58059fa17c8e360.tar.gz"
        ],
        sha256=("1cfe0ebde2c6dd80405977e0b3a6f72e1b062d8a79f9f0437ebebe463c9c85f7"),
        strip_prefix="mujoco_menagerie-469893211c41d5da9c314f5ab58059fa17c8e360/",
    ),
)
original_file = package_map.ResolveUrl(
    "package://mujoco_menagerie/aloha/scene.xml"
)
drake_compatible_file = original_file.replace(".xml", ".drake.xml")
MakeDrakeCompatibleModel(original_file, drake_compatible_file)
visualizer.AddModels(filename=drake_compatible_file)
visualizer.Run(loop_once=not running_as_notebook)
```
which failed with `RuntimeError: Desired state input port for model instance aloha_scene not connected.`

I'm not sure which component `ModelVisualizer` lands in today. I've tagged @rpoyner-tri for parsing, but please delegate as needed.

My inclination is that the right solution might be to make the desired state input port of the plant support being disconnected.

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
RuntimeError                              Traceback (most recent call last)
Cell In[8], line 28
     26 MakeDrakeCompatibleModel(original_file, drake_compatible_file)
     27 visualizer.AddModels(filename=drake_compatible_file)
---> 28 visualizer.Run(loop_once=not running_as_notebook)
     29 meshcat.DeleteAddedControls()

File ~/manipulation/venv/lib/python3.10/site-packages/pydrake/visualization/_model_visualizer.py:511, in ModelVisualizer.Run(self, position, loop_once)
    496 """
    497 Runs the model. If Finalize() hasn't already been explicitly called
    498 then the object will be finalized first.
   (...)
    508   loop_once: a flag that exits the evaluation loop after one pass.
    509 """
    510 if self._builder is not None:
--> 511     self.Finalize(position=position)
    512 else:
    513     self._check_rep(finalized=True)

File ~/manipulation/venv/lib/python3.10/site-packages/pydrake/visualization/_model_visualizer.py:406, in ModelVisualizer.Finalize(self, position)
    402     self._sliders.SetPositions(position)
    404 # Use Simulator to dispatch initialization events.
    405 # TODO(eric.cousineau): Simplify as part of #13776 (was #10015).
--> 406 Simulator(self._diagram).Initialize()
    407 # Publish draw messages with current state.
    408 self._diagram.ForcedPublish(self._context)

RuntimeError: Desired state input port for model instance aloha_scene not connected.
```
```

amcastro-tri, 2025-01-13:
`kLagged` is the default approximation @RussTedrake.
If you remove `plant.set_discrete_contact_approximation(DiscreteContactApproximation.kLagged)` your reproduction no longer throws? Could you confirm?

`kLagged` became the default about a mont ago with [#21294](https://github.com/RobotLocomotion/drake/pull/21294)

jwnimmer-tri, 2025-01-13:
It's not a problem with lagged.  It's a problem with the disconnected input port during visualization.

amcastro-tri, 2025-01-14:
I imagine that no matter where the model visualizer is grabbing its defaults in Russ's example, we'd still like to update the requirement to be connected for the desired state input ports.
Let me know if you agree with this solution:

>  If the desired state input port is not connected, that PD controller is ignored/not added to the model.
> With the corresponding doc updates.

If you agree, I can work on this.

Also, fyi, I just realized our docs are not correct. We document [actuation inputs to be required](https://github.com/RobotLocomotion/drake/blob/e9fefcbe289240941a380af3ec189fee5ae1fb87/multibody/plant/multibody_plant.h#L645) when there are no PD controllers. That's not true. I'll also update those docs in the same PR.

RussTedrake, 2025-01-14:
> If the desired state input port is not connected, that PD controller is ignored/not added to the model.

This sounds like the right solution to me. Thank you. Should we reassign the issue to you?

nit: "ignored/not added to the model" might be a little confusing. I believe the behavior will be that if I build the model with PD gains and run with the desired state disconnected, then the internal PD has no effect. However, if I were to then connect the port and step, it _would_ take effect. So it _was_ added to the model, but was effectively deactivated?

jwnimmer-tri, 2025-01-14:
"deactivated" is good.  Another option would be "the controller is disarmed when no input is connected".

jwnimmer-tri, 2025-02-27:
Inside TRI, our bestiary of robot species is growing fast, nearly all of them are going to use implicit PD, and it's super annoying to keep hitting this bug.  Bumping up to high priority.

PR #22542 (2025-03-06): Desired state input ports are no longer required to be connected
Closes #22444.

Per discussion in #22444, we allow the desired state input ports for a model instance to be disconneted. This "disarms" PD control for such model instance, meaning that this controller won't have any effect on the dynamics.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22542)
closed 2025-03-06 by jwnimmer-tri