RobotLocomotion/drake #22460: `cmake: cannot execute binary file` in mac build

ggould-tri, 2025-01-14:
### What happened?

A CI build failed with the error:
```
[3:16:20 AM]  /bin/sh: /opt/homebrew/bin/cmake: cannot execute binary file
[3:16:20 AM]  make[1]: *** [CMakeFiles/drake_cxx_python.dir/all] Error 126
[3:16:20 AM]  make: *** [all] Error 2
```

This is described in slack as having happened twice before: 
 https://drakedevelopers.slack.com/archives/C270MN28G/p1736782108393449

This is not the first occurrence, so it stays open.

### Version

current HEAD

### What operating system are you using?

macOS 15 (Sequoia)

### What installation option are you using?

compiled from source code using CMake

### Relevant log output

```shell
[3:16:35 AM]  INFO: Found 1 target...
[3:16:35 AM]  Target @@drake+//:install up-to-date:
[3:16:35 AM]    /Users/admin/workspace/mac-arm-sequoia-clang-cmake-nightly-everything-release/_cmake_admin/_bazel_admin/d012f0c58e499e449def8b9401503fed/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/drake+/install_test_commands
[3:16:35 AM]    /Users/admin/workspace/mac-arm-sequoia-clang-cmake-nightly-everything-release/_cmake_admin/_bazel_admin/d012f0c58e499e449def8b9401503fed/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/drake+/install
[3:16:35 AM]  [10,058 / 10,058] no actions running
[3:16:36 AM]  INFO: Elapsed time: 1463.163s, Critical Path: 152.00s
[3:16:36 AM]  INFO: 10058 processes: 2 remote cache hit, 5131 internal, 4920 darwin-sandbox, 5 worker.
[3:16:36 AM]  INFO: Build completed successfully, 10058 total actions
[3:16:36 AM]  /bin/sh: /opt/homebrew/bin/cmake: cannot execute binary file
[3:16:36 AM]  make[1]: *** [CMakeFiles/drake_cxx_python.dir/all] Error 126
[3:16:36 AM]  make: *** [all] Error 2
[3:16:36 AM]  Error(s) when building project
[3:16:36 AM]     0 Compiler errors
[3:16:36 AM]     4 Compiler warnings
[3:16:39 AM]  -- *** Submitted to CDash with build id 1830504
[3:16:39 AM]  
[3:16:39 AM]    ------------------------------------------------------------------------------
[3:16:39 AM]    *** Disk usage for /tmp:
[3:16:39 AM]   Filesystem      Size    Used   Avail Capacity iused ifree %iused  Mounted on
[3:16:39 AM]  /dev/disk2s5   186Gi    23Gi   147Gi    14%    576k  1.5G    0%   /System/Volumes/Data
[3:16:39 AM]    ------------------------------------------------------------------------------
[3:16:39 AM]  
[3:16:39 AM]  
[3:16:39 AM]    ------------------------------------------------------------------------------
[3:16:39 AM]    *** CTest Result: FAILURE DURING CMAKE BUILD
[3:16:39 AM]    ------------------------------------------------------------------------------
[3:16:39 AM]    *** CDash URL: https://drake-cdash.csail.mit.edu/index.php?project=Drake&showfilters=1&filtercount=2&showfilters=1&filtercombine=and&field1=label&compare1=61&value1=jenkins-mac-arm-sequoia-clang-cmake-nightly-everything-release-24&field2=buildstarttime&compare2=84&value2=now
[3:16:39 AM]    ------------------------------------------------------------------------------
[3:16:39 AM]  
[3:16:39 AM]  CMake Error at /Users/admin/workspace/mac-arm-sequoia-clang-cmake-nightly-everything-release/ci/ctest_driver_script.cmake:154 (message):
[3:16:39 AM]    *** Return value set to NON-ZERO due to failure during build
[3:16:39 AM]
```

BetsyMcPhail, 2025-01-17:
1/17 - [mac-arm-sequoia-clang-cmake-nightly-release](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/32)

BetsyMcPhail, 2025-01-20:
1/19 - https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/32
1/20 - https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/33

williamjallen, 2025-01-21:
1/21 - https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-release/36/consoleFull

ggould-tri, 2025-01-21:
With this now producing daily failures, I am going to put it up to P2.

williamjallen, 2025-01-25:
1/25 - https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-release/41/consoleFull

williamjallen, 2025-01-28:
1/28 - https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-release/44/consoleFull

SeanCurtis-TRI, 2025-01-31:
1/31/ - https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/45/

ggould-tri, 2025-03-10:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/92/console

BetsyMcPhail, 2025-03-13:
3/13 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/94/

jwnimmer-tri, 2025-03-24:
If we don't see this again in the next ~1 week, we'll close this as "seems okay now".

closed 2025-04-01 by tyler-yankee