RobotLocomotion/drake #22476: Investigate major version upgrade of "SSH Build Agents" Jenkins plugin

williamjallen, 2025-01-16:
The [SSH Build Agents](https://plugins.jenkins.io/ssh-slaves/) Jenkins plugin released a new version which Jenkins indicates is not backwards compatible.  This issue serves as a placeholder to remind us to investigate what it takes to upgrade this plugin.

[image]

tyler-yankee, 2025-03-12:
Sure, happy to take a look. (Though I haven't been into Jenkins' plugins updates much, so I may have questions.)

williamjallen, 2025-03-13:
@tyler-yankee that seems reasonable to me.  Perhaps we can give it a try tomorrow?

tyler-yankee, 2025-03-13:
@williamjallen: Sounds good to me.

tyler-yankee, 2025-03-17:
Successfully updated, and nothing with SSH agents seems to be on fire.

closed 2025-03-17 by tyler-yankee