RobotLocomotion/drake #22530: Top-to-bottom review of installation documentation

jwnimmer-tri, 2025-01-25:
For people getting started with Drake, we have an installation page here:
- https://drake.mit.edu/installation.html

It links to sub-pages:
- https://drake.mit.edu/pip.html
- https://drake.mit.edu/apt.html
- https://drake.mit.edu/from_binary.html
- https://drake.mit.edu/docker.html

And related pages:
- https://drake.mit.edu/troubleshooting.html
- https://drake.mit.edu/python_bindings.html

+@BetsyMcPhail I'd like to ask you to put on your tech writer hat and take a full pass over all of those pages, to make improvements or at least identify areas for improvement and we can discuss what changes to make.

(We can sync up f2f on this in our next meeting.)

jwnimmer-tri, 2025-02-10:
My general interest is making it better _organized_.

Also we might want to improve:
- https://github.com/RobotLocomotion/drake?tab=readme-ov-file#drake
- https://drake.mit.edu/

BetsyMcPhail, 2025-02-17:
While reviewing the documentation, I noticed that [https://hub.docker.com/r/robotlocomotion/drake](https://hub.docker.com/r/robotlocomotion/drake) was out-of-date. I removed all references to focal, added noble and moved `latest` to refer to jammy.

tyler-yankee, 2025-02-25:
@jwnimmer-tri: @BetsyMcPhail and I have discussed some aspects of the pages that could be cleared up and/or reorganized which are all relatively straightforward. We had one more major suggestion that we wanted to ask you about.

In general, it seems that the installation instruction pages require jumping a lot from page to page and going through many links. For example, if as a developer I want to access the documentation for using Bazel, I have to start from the Installation page, find the link to the Source Installation page, and then scroll to the bottom to get to the link for the Bazel page. I feel that what would make this workflow much more user-friendly is adding links to all subpages when you hover over "Installation" in the top navigation bar, similar to "API Documentation" and "Resources." It would list the pages you've linked above (pip, apt, binary, docker) as well as source; that list could change as I dive into the issue more, but from what I understand right now that seems sufficient. Your thoughts?

jwnimmer-tri, 2025-02-25:
Which kind of person/role do you have in mind where that improvement would help?

For the Bazel case in particular you mention, the https://drake.mit.edu/from_source.html#building-with-bazel section is so small and trivial as not to be worth putting into the menu bar.  Or if the user is a drake developer looking for that information, then adding it to the "Installation" menu isn't going to help, instead we should make the [Resources > For Developers](https://drake.mit.edu/developers.html) menu target easier to use / better organized.   But that's out of scope for this issue.   Bazel doesn't install anything, so anything Bazel-related is not relevant here.

On the other hand, I could easily believe that an end user who comes in already expecting to use `pip` would be helped out by an `Installation > Pip` menu that goes directly to https://drake.mit.edu/pip.html.  The only thing we need to be careful of when sending them directly to that page is that the might miss out on some of the important background information at https://drake.mit.edu/installation.html like which Python versions are supported.  There is a backlink to that from the pip page, but the median user only click through to a recommended links after they fail, so many will skip over it.  We might need to improve the pip page introductory material to be more standalone, possibly repeating information from other pages.

tyler-yankee, 2025-02-25:
I should have provided a link; I was referring to https://drake.mit.edu/bazel.html which is linked under "Making changes to Drake". I guess this is a separate issue: this page doesn't necessarily relate to installation, but it also seems generally difficult to access from elsewhere on the website, unless I'm missing something (very possible!). I agree that the "Building with Bazel" section is trivially small to include a menu bar link for.

I agree with making the pip and other pages more standalone to include the introductory information from the top-level page.

jwnimmer-tri, 2025-02-25:
> ... I feel that what would make this workflow much more user-friendly is adding links to all subpages when you hover over "Installation" in the top navigation bar, similar to "API Documentation" and "Resources." ...

BTW assuming we add those pages to the menubar, we do still need to give people a menu link directly to `installation.html`.  So maybe the first submenu item is named something like "Overview" and links to `installation.html`.

> ... but it also seems generally difficult to access from elsewhere on the website ...

Yes.  The page for https://drake.mit.edu/developers.html should link to `bazel.html`.

jwnimmer-tri, 2025-04-03:
Fixed in #22746, I think.

closed 2025-04-03 by jwnimmer-tri