RobotLocomotion/drake #22532: Add placeholder variables workflow to KinematicTrajectoryOptimization

RussTedrake, 2025-01-26:
Most of our trajectory optimization workflows use this; it allows us to e.g. add constraints at a particular time by binding a constraint to "placeholder variables" for q, qdot, qddot, etc and automatically substituting them for the necessary combinations of control points behind the API.

As discussed w/ @hongkai-dai and (separately) @cohnt and @shaoyuancc .

hongkai-dai, 2025-01-26:
I have a working branch on this feature. I will get back to the feature when I am back from China

RussTedrake, 2025-01-26:
Great!  Thanks!

hongkai-dai, 2025-01-29:
When we have this placeholder variable for `q` and `qdot`, do we plan to replace the existing functions in KinematicTrajectoryOptimization to use the placeholder variable? I mean the function

- AddPathPositionConstraint(lb, ub, s) implements lb <= q(s) <= ub.
- AddPathPositionConstraint(constraint, s) implements lb <= constraint(q(s)) <= ub
- AddVelocityConstraintAtNormalizedTime(constraint, s) implements lb <= constraint(q(s), qdot(s)) <= ub

It seems to me that these three constraints can be replaced by 
```
AddConstraint(constraint, placeholder_vars)
```
where `placeholder_vars` include the placeholder variables for q and qdot.

RussTedrake, 2025-02-01:
It's a good question. The only argument I see for keeping them is that we do have a lot of constraints, e.g. in the inverse_kinematics folder (like [PositionConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_position_constraint.html)), that explicitly take the vector `q` as decision variables, and a few (like [SpatialVelocityConstraint](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_spatial_velocity_constraint.html)) that assume `[q,v]`.  For that reason, I would perhaps perfer improving the documentation on that method (and even PositionConstraint, etc, which isn't as clear as some of the others what variables it should be bound to) and leave them?

The placeholder variables will be very valuable for the new constraint you're adding and others.

PR #22599 (2025-02-12): KinematicTrajectoryOptimization adds velocity linear constraint.
Use the placeholder variable in the constraint binding. Adds a linear constraint on the decision variable when the user wants lb <= A * qdot(s * T) <= ub.

Also resolves #22532 

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22599)
closed 2025-02-12 by hongkai-dai