RobotLocomotion/drake #22539: Switch CDash DB backend from MySQL to Postgres

williamjallen, 2025-01-27:
Kitware is planning to drop support for MySQL as a CDash database backend to make better use of Postgres-specific features.  This issue is a placeholder for the task of migrating our existing Aurora MySQL instance to Aurora Postgres.

williamjallen, 2025-02-11:
@jwnimmer-tri Re: the "reset the CDash database" idea, it looks like Drake's CDash instance is configured to retain 5 months of data.

jwnimmer-tri, 2025-02-11:
I imagine we're not the only user of CDash that needs to migrate from Aurora to Aurora.  Can you explain in the issue here what the challenge is?  I know we talked about it in the meeting yesterday but it didn't make it into the notes.

jwnimmer-tri, 2025-02-24:
From f2f -- the issue with cost was a misunderstanding, not actually relevant.  So the only work here is moving the database via a userland tool like `pgloader`.

williamjallen, 2025-06-08:
06/07: CDash has been migrated to Postgres in conjunction with an update to CDash 4.0.  Closing as complete.