RobotLocomotion/drake #22543: InverseKinematics throws when models have a QuaternionFloatingJoint (for a non-floating body)

marip8, 2025-01-27:
### What happened?

When adding a `QuaternionFloatingJoint` between two bodies where one is not the world body, `InverseKinematics` throws an exception (see error message below). I think this is effectively the same issue as #21215, but with the `InverseKinematics` class. Per [this comment](https://github.com/RobotLocomotion/drake/issues/21215#issuecomment-2616943300), however, the solution in #21308 does not fix this issue in `InverseKinematics`

### Version

v1.36

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

pip install drake

### Relevant log output

```shell
Failure at bazel-out/k8-opt/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/rigid_body.h:324 in floating_positions_start(): condition 'is_floating()' failed.
```

jwnimmer-tri, 2025-01-28:
Is there a simple way to reproduce the problem, perhaps a small Python program?

marip8, 2025-01-28:
```python
from pydrake.all import RobotDiagramBuilder, InverseKinematics, QuaternionFloatingJoint

def main():
    model_sdf = """
    <?xml version="1.0" ?>
    <sdf version="1.7">
      <world name="test">
        <include>
          <uri>package://drake_models/iiwa_description/sdf/iiwa14_polytope_collision.sdf</uri>
          <name>robot</name>
        </include>
        <joint name="world_to_robot" type="fixed">
          <parent>world</parent>
          <child>robot::iiwa_link_0</child>
        </joint>
    
        <model name="object">
          <pose>1.0 0.0 0.5 0 0 0</pose>
          <link name="base_link">
            <visual name="visual">
              <geometry>
                <box>
                  <size>0.320 0.200 0.300</size>
                </box>
              </geometry>
            </visual>
          </link>
        </model>
    
        <frame name="tcp" attached_to="robot::iiwa_link_7">
          <pose relative_to="robot::iiwa_link_7">0 0 0.045 0 0 0</pose>
        </frame>
      </world>
    </sdf>
    """

    rdb = RobotDiagramBuilder(time_step=0)
    parser = rdb.parser()
    parser.AddModelsFromString(file_contents=model_sdf, file_type='sdf')

    robot_model = rdb.plant().GetModelInstanceByName('robot')
    object_model = rdb.plant().GetModelInstanceByName('object')

    floating_joint = rdb.plant().AddJoint(
        QuaternionFloatingJoint(
            'floating_joint',
            rdb.plant().GetFrameByName('tcp', robot_model),
            rdb.plant().GetFrameByName('base_link', object_model)
        )
    )

    rdb.plant().Finalize()

    ik = InverseKinematics(rdb.plant())
    pass

if __name__ == '__main__':
    main()

```

rpoyner-tri, 2025-01-28:
assigning @hongkai-dai for further assessment and delegation.

hongkai-dai, 2025-02-01:
I think Drake's thrown error is meaningful and correct here. You tried to add a floating joint between the base frame and a frame (`tcp` frame) rigidly attached to link 7, what's the meaning of this? Neither the base frame or the tcp frame are floating, why adding a floating joint here?

jwnimmer-tri, 2025-02-01:
> ... why adding a floating joint here?

Take a look at the mentioned issue #21215.  The extra joint is to enable conditionally joint locking the manipuland to the gripper, to simulate holding it in a grasp.

It seems the logic I added in #19442 which was recently relocated in #22361 related to floating bodies is not entirely correct.  It assumes that the way to identify quaternion dofs is via bodies, but I think the same as in #21308 we should change the loop to be over joints, instead, now that all positions are always covered by a joint.

PR #22578 (2025-02-01): [ik] Allow non-world floating joints without crashing
Also fix missing initial guess for locked non-quaternion dofs.

Closes #22543.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22578)
closed 2025-02-01 by jwnimmer-tri