RobotLocomotion/drake #22547: CMake options for solver opt-out

jwnimmer-tri, 2025-01-28:
**Is your feature request related to a problem? Please describe.**

#22393 added flags for downstream Bazel users to opt-out of our open-source solvers (e.g., Clarabel).  However, the need to opt-out applies equally well to downstream CMake users who are building and installing Drake from source.

**Describe the solution you'd like**

Offer CMake options to opt-out of the same solvers as the Bazel flags.  (This should be as simple as tweaking CMake's generated bazelrc file to set the bazel flag when needed.

Update the documentation at https://drake.mit.edu/from_source.html to explain the new options.

**Describe alternatives you've considered**

Users could make a magical `user.bazelrc` file to set the Bazel flag during a CMake build, but that's very awkward.

**Additional context**

N/A

nicolecheetham, 2025-02-03:
@jwnimmer-tri This issue mentions a simple tweak is needed in cmake's generated bazelrc file. I wanted to confirm if the desired change is `build:{solver_name} --@drake//tools/flags:with_{solver_name}=True` for each open source solver after line 41 in drake/cmake/bazel.rc.in.

I also wanted to check for testing the options with drake_cmake_external, I would create the specific option(s) then pass them into ExternalProject_Add(drake_external_examples) via CMAKE_ARGS. Then, I would build locally to check if it build properly.

On another note, is there any additional bazel documentaion or resources to better understand the workings of bazel.rc files? I just found https://bazel.build/run/bazelrc while looking over 22393.

jwnimmer-tri, 2025-02-03:
Best guess, the `drake/CMakeLists.txt` snippet would be along these lines, with no changes to `bazel.rc.in`:

```cmake
option(WITH_CLARABEL "Build with support for Clarabel" ON)

if(NOT WITH_CLARABEL)
  string(APPEND BAZEL_CONFIG " --@drake//tools/flags:with_clarabel=False")
endif()
```

> I also wanted to check for testing the options with drake_cmake_external,  ....

I don't know.  Kitware are the CMake exports, hopefully you can find the answer in-house.

> ... documentaion or resources to better understand the workings of bazel.rc files?

The link you have is the best one about the files themselves.  I think it's pretty thorough.

In terms of what flags to what, that's a question of whatever the bazel command line offers (rcfiles just tack things onto the command line) so https://bazel.build/reference/command-line-reference and and https://bazel.build/reference/be/general#config_setting.

tyler-yankee, 2025-02-12:
@jwnimmer-tri: It's unclear to me how drake-external-examples should be updated. The flags for open-source solvers in [drake/tools/flags](https://github.com/RobotLocomotion/drake/blob/6d0d71c308a369571d5b46e5c70af8a0cd3df894/tools/flags/BUILD.bazel#L230) describes that we can't guarantee tests passing if one of the solvers is disabled. I understand wanting to test (and demonstrate) the CMake option, but I don't want to make a test here flake either?

An alternative to just updating the call to `ExternalProject_Add(drake` would be to provide an option to the user to disable the solver, and then point them in the README to the code that would disable it when they enable that option.

jwnimmer-tri, 2025-02-12:
> The flags for open-source solvers in drake/tools/flags describes that we can't guarantee tests passing if one of the solvers is disabled.

That's talking about the full suite of Drake's own tests.  The DEE examples use the solvers mostly not at all, so will not be a hazard.  Something like `WITH_CSDP=OFF` should be safe, and then a DEE test can check that `CsdpSolver().available()` returns false.

(If we don't want to test via DEE that's fine too, but we need to test _somewhere_.)

jwnimmer-tri, 2025-02-25:
Finished as of the two linked PRs.

closed 2025-02-25 by jwnimmer-tri