RobotLocomotion/drake #22574: [ci] Running drake-external-examples jobs against Drake pull requests

jwnimmer-tri, 2025-01-31:
**Is your feature request related to a problem? Please describe.**

Our code in "DEE" (https://github.com/RobotLocomotion/drake-external-examples/) not only provides Drake users with documentation about how to use Drake downstream, but also provides Drake developers with test coverage that Drake works correctly when used downstream -- that our binary packages can install and run, that our CMake interface can build, that our bzlmod-exported dependency structure resolves correctly, etc.

When a Drake PR is changing build infrastructure, it risk breaking some of those guarantees.  Developers can manually run most of the DEE tests locally, but not always all of them (e.g., macOS) and anyway running locally is a time-consuming and manual process, so often is done less than it should be. This leads to bugs slipping through to post-merge jobs (i.e., nightly), and hotfixes the next day.  It also means that sometimes are nightly packaging builds we ship to users are broken.

**Describe the solution you'd like**

Add documentation to https://drake.mit.edu/jenkins.html that explains how to launch DEE job(s) again a pull request, on an opt-in basis.

All DEE flavors (currently N=8) must be supported, but need not all happen under the same umbrella -- it's okay to require the developer to click on multiple pages to get things launched.  Of course all else being equal fewer clicks are better, but since this exercise happens somewhat rarely, we don't need to optimize it; we just need it to exist in some reasonable form.

In the best case, launching the job is a 'bot comment like with our normal provisioned / unprovisioned jobs; this is very simple and already understood.  It's also fair to ask the developer to open a web page and copy-paste some git shas and click some buttons.  Another valid but slightly annoying option would be to ask the developer to open a dummy DEE PR, and somehow point that PR to test the Drake PR in the code itself.  The only mandatory requirement is that the job runs in CI, not locally.  Whatever we can do to make that easy is better, but the request here is only to make it _possible_.

**Describe alternatives you've considered**

N/A

**Additional context**

N/A

tyler-yankee, 2025-02-14:
Related, the [Drake Jenkins documentation](https://drake.mit.edu/jenkins.html) currently says that the following is an option to trigger a build:

> @drake-jenkins-bot ok to test to accept this pull request for testing.

I don't see this is any of the drake jenkins-job-builder code. Am I missing something, or is this out of date?

jwnimmer-tri, 2025-02-14:
It's still true.  I believe it's a setting in Jenkins GHPRB itself.

tyler-yankee, 2025-02-17:
Pinging @betsymcphail and @jwnimmer-tri for thoughts on how to proceed with this feature. I'd like to find a balance between not needing to completely overhaul CI while also providing something useful and easy-to-use for Drake developers.

In order to support this request, at minimum the CI for each example in DEE will need some sort of parameterization to be able to use a PR branch of Drake, instead of (master)(nightly release)(etc.). This will likely involve passing some flags to change URLs in either Bazel or CMake, and using the experimental wheels for the PyPI examples (I've verified the approach for both of these locally).

From there, there is GitHub Actions and Jenkins to deal with:

* For GHA, I think the most straightforward option is for developers to manually trigger builds using a [workflow dispatch](https://github.blog/changelog/2020-07-06-github-actions-manual-triggers-with-workflow_dispatch/). Looks like we could have developers input their PR branch name and a Git SHA as parameters through GitHub's UI. We can make individual workflows for each example, so that developers only build the examples that they need.
* For Jenkins, it *should* be possible to integrate with the existing drake-jenkins-bot and trigger builds from the Drake PR. We would need additional code inside drake-jenkins-jobs that uses the Git environment variables and the new parameterized DEE CI jobs to provide new jobs which build each individual example (again, developers building only what they need). Note that this means the Jenkins job code for DEE on Drake PRs would live separate from the Jenkinsfile inside DEE itself...but I feel like that makes sense, because the impetus for triggering these builds is from Drake, not DEE.

Finally, the bots (for drake-jenkins-bot, but also includes CDash and Girder since they are grouped together in GitHub currently) may need admin access to DEE for the Jenkins stuff to work.

jwnimmer-tri, 2025-02-18:
> ... developers to manually trigger builds using a workflow dispatch.  Looks like we could have developers input their PR branch name and a Git SHA as parameters through GitHub's UI. 

Sounds fine.

> We can make individual workflows for each example, so that developers only build the examples that they need.

Ideally the developer would just input their branch name once.  More clicks is worse.  It's fine if it runs all of the examples, it's probable that's what they want anyway.

> Note that this means the Jenkins job code for DEE on Drake PRs would live separate from the Jenkinsfile inside DEE itself ...

This sounds fine.  We want to avoid copy-paste, but by design all of the important build logic is in bash scripts not Jenkinsfile, so having a different Jenkinsfile just for the triggering part seems okay.

tyler-yankee, 2025-02-19:
Sounds good to me.

Is it fair to ask developers to run the respective experimental packaging jobs and input the URL for those, in order to run the `drake_bazel_download` and `drake_cmake_installed` examples? Otherwise, we may not as well implement the feature for these examples at all, since it will somehow involve downloading and building source which is covered by the others.

jwnimmer-tri, 2025-02-19:
> Is it fair to ask developers to run the respective experimental packaging jobs and input the URL from AWS ...

Yes; we can document the steps in the https://drake.mit.edu/jenkins.html section about it.

tyler-yankee, 2025-03-04:
> we can document the steps in the https://drake.mit.edu/jenkins.html section about it.

My only nit-pick here is that we would be documenting steps to run GitHub Actions on a page called "jenkins." Do we rename the page title and/or URL to be more general?

jwnimmer-tri, 2025-03-04:
If you prefer, I would be OK with a new GHA page linked from https://drake.mit.edu/developers.html#continuous-integration-notes as a sibling to the Jenkins page.

I would also be OK editing the title of the existing Jenkins page a bit to make it more inclusive.

I would prefer NOT to rename the file `jenkins.html` since a lot of people have that bookmarked.

BetsyMcPhail, 2025-03-12:
> Is there a preference on where this job lives in Jenkins? I was thinking of calling it "linux-jammy-unprovisioned-external-examples," but please let me know if you'd like a different name.
> 
> It could be listed in the Pull Requests view, or some new "External Examples" view, or just float with no view.

`linux-jammy-unprovisioned-external-examples` sounds reasonable. I would prefer not in Pull Requests, as those jobs run automatically on PRs. A new view would be easy for users to find.

tyler-yankee, 2025-03-28:
It seems that everything here is finally ready to go. I believe the merge order should be:

- [DEE #374](https://github.com/RobotLocomotion/drake-external-examples/pull/374) GHA example parameterization
- [DEE #376](https://github.com/RobotLocomotion/drake-external-examples/pull/376) Jenkins example parameterization
- [DJJ #161](https://github.com/RobotLocomotion/drake-jenkins-jobs/pull/161) Jenkins job (and view) definition
- [Drake #22721](https://github.com/RobotLocomotion/drake/pull/22721) documentation

closed 2025-04-14 by BetsyMcPhail