RobotLocomotion/drake #22581: Upgrade macOS to Python 3.13 to match Homebrew

jwnimmer-tri, 2025-02-02:
**Is your feature request related to a problem? Please describe.**

The default `python@3` in Homebrew is `python@3.13` now.  Due to #21968 we have been unable to upgrade to it, but soon that issue will be fixed.  Using a non-default version is annoying for our users.

**Describe the solution you'd like**

Once that blocking issue is fixed, we should update our support matrix and packaging scripts to add / use `python@3.13`.

For https://drake.mit.edu/from_binary.html, that means _changing_ our binaries to use 3.13 instead of 3.12.  This will be a change to our `setup/mac/...` scripts.

For https://drake.mit.edu/pip.html, we will start by adding a 3.13 wheel.  In our next release (v1.38.0, mid-February) I will announce the mac 3.11 wheels as deprecated, and once the deprecation has timed out we'll stop publishing the mac 3.11 wheels.

**Additional context**

The prior issue (3.11 => 3.12) was #20294.

jwnimmer-tri, 2025-02-24:
FYI @BetsyMcPhail I checked and this issue is ready to go, once the blocking prior issue finishes.

tyler-yankee, 2025-02-27:
Noting here, whether as part of this issue or as a follow-up issue, that once this is done DEE can also be updated to use Python 3.13 on macOS in its CI jobs for both `drake_pip` and `drake_poetry`. It may still be useful to allow the setup scripts to take a `--python-version` argument, since that is more general (even though that logic was originally only added to allow macOS to use 3.12 instead of the default 3.13).

jwnimmer-tri, 2025-03-10:
FYI @BetsyMcPhail https://github.com/RobotLocomotion/drake/issues/21968#issuecomment-2711823171.  So far, it seems not _quite_ ready to start working on this.  I'll post back once it is.

jwnimmer-tri, 2025-03-11:
Okay, folks, I believe this should be unblocked now.

If you start from my [python-313-wip](https://github.com/jwnimmer-tri/drake/tree/python-313-wip) tree as a baseline, I hope that the Homebrew Python 3.13 interpreter will be compatible with Drake.  Eventually we'll have #22580 land on master, but until then you'll just need to test using your own branches based on my tree.  We won't be able to land your PRs on master until my PR lands first, but you should still be able to explore starting now.

Most of the work will be changing the https://github.com/RobotLocomotion/drake/tree/master/setup/mac prereqs code, and the various other places that 3.12 is hard-coded in Drake for macOS.  Also the documentation that explains our supported versions, and the https://drake.mit.edu/release_notes/end_of_support.html page should listing the change in macOS packages.  Also Tyler's note above.

jwnimmer-tri, 2025-03-14:
> Okay, folks, I believe this should be unblocked now.

All prereqs have landed now.  This should be unblocked to develop and land on master.

The pending #22580 is related for linux, but shouldn't be required for any macOS changes.

jwnimmer-tri, 2025-04-11:
I think possibly this is finished now?  Or did you want to wait for DEE 380 downstream upgrades as part of this ticket, too?

tyler-yankee, 2025-04-11:
Yeah, I was waiting until the DEE updates land - if you think it makes more sense to make a separate ticket on DEE for that, I don't mind either way.

closed 2025-04-14 by BetsyMcPhail