RobotLocomotion/drake #22582: CameraConfig.background yaml doesn't successfully parse in python

SeanCurtis-TRI, 2025-02-03:
### What happened?

The `CameraConfig` structure has a `geometry::Rgba`-valued member called `background`.

As outlined in [this stackoverflow](https://stackoverflow.com/questions/79398649/specify-background-color-in-scenario-camera-config-yaml/79400212?noredirect=1#comment140027250_79400212) it can't really be set. The `!Rgba` token fails and setting it directly with a list of 3-4 floats also doesn't work.  For example:

```py
        yaml = "background: [0.1, 0.2, 0.3, 0.4]"
        dut = yaml_load_typed(schema=mut.CameraConfig, data=yaml)
```

Produces the error message (same as reported in stackoverflow):
```
Traceback (most recent call last):
  File "/mnt/nobackup/cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/_main/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/sensors_test.runfiles/_main/bindings/pydrake/systems/test/sensors_test.py", line 262, in test_camera_config_yaml
    dut = yaml_load_typed(schema=mut.CameraConfig, data=yaml)
  File "/mnt/nobackup/cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/_main/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/sensors_test.runfiles/_main/bindings/pydrake/common/yaml.py", line 648, in yaml_load_typed
    _merge_yaml_dict_into_target(
  File "/mnt/nobackup/cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/_main/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/sensors_test.runfiles/_main/bindings/pydrake/common/yaml.py", line 567, in _merge_yaml_dict_into_target
    _merge_yaml_dict_item_into_target(
  File "/mnt/nobackup/cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/_main/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/sensors_test.runfiles/_main/bindings/pydrake/common/yaml.py", line 515, in _merge_yaml_dict_item_into_target
    _merge_yaml_dict_into_target(
  File "/mnt/nobackup/cache/bazel/_bazel_seancurtis/c482a09839de82c7b8c8a508d26ab2c3/execroot/_main/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/sensors_test.runfiles/_main/bindings/pydrake/common/yaml.py", line 528, in _merge_yaml_dict_into_target
    assert isinstance(yaml_dict, collections.abc.Mapping), yaml_dict
AssertionError: [0.1, 0.2, 0.3, 0.4]
```

### Relevant log output

```shell

```

SeanCurtis-TRI, 2025-02-03:
The following yaml is erroneous:

```yaml
background: [0.1, 0.2, 0.3, 0.4]
```

The correct spelling would be:

```yaml
background:
  rgba: [0.1, 0.2, 0.3, 0.4]
```

So, the defect is not that there isn't a yaml spelling that gets parsed. It's that the spelling is incredibly arcane.

This is certainly sufficient to parse but we don't have any concrete examples or documentation -- at least, none that I've found yet.

SeanCurtis-TRI, 2025-02-03:
I believe [the documentation](https://github.com/RobotLocomotion/drake/blob/master/geometry/rgba.h#L94-L108) on `Rgba` is insufficient. The key `rgba` seems like a metavariable. I.e., I've got some `Rgba`-valued field (e.g., `background`) and I set its value as a simple list of values. In reality, the *value* of `background` is a sub dictionary that *requires* the `rgba` key -- that is the glyph that allows the list to be converted to `Rgba`.

Recommendations:

1. Clean up the `Rgba` documentation to be more clear.
2. In the yaml documentation, we should call out a list of "weird" cases which would include, at the very least, `Rgba`. It should probably also include `MemoryFile` and other structs that have a non-obvious yaml mapping.

jwnimmer-tri, 2025-02-03:
> In the yaml documentation, we should call out a list of "weird" cases which would include, ...

I don't think we want global variables in our documentation.  Each class's overview (Rgba, etc.) should explain its YAML syntax, and give concrete examples.  Having a Drake-wide compendium of per-class YAML details will not scale.

SeanCurtis-TRI, 2025-02-03:
I didn't mean that the global documentation would spell out the weirdness. But there should be a list of classes that are weird and it should link to that class's weirdness.

The problem I'm trying to address is the discovery of the weirdness.

Admittedly, the problem is already "solved". If I'm trying to parse something of type `Rgba`, I should probably run to see if the class says anything. So, to that extent, some kind of glossary of weirdness would be strictly redundant in the name of an alternate entry point. So, based on that, I'm open to the idea that maintaining such a glossary would be more work than it's worth.

nepfaff, 2025-02-04:
Oh, I only checked the `Detailed Description` section of `Rgba` and thus missed the documentation under `Serialize`. That seems like a good place for it, though. I will know next time

PR #22584 (2025-02-05): Tweak some yaml serialization documentation
1. MemoryFile's documentation choked doxygen. There was no documentation rendered for Serialize(). The minimum fix led to really ugly layout. (Code block immediately followed indented block.) So, we shuffled the rhetoric so the layout looked better.
2. Rgba suffered from a similar layout problem. Furthermore, it wasn't clear that it had particularly weird yaml semantics. The documentation has been extended to help that become clearer.

closes #22582

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22584)
closed 2025-02-05 by jwnimmer-tri