RobotLocomotion/drake #22587: macOS Sequoia: `file does not start with MH_MAGIC[_64]`

tyler-yankee, 2025-02-04:
### What happened?

Nightly builds on macOS Sequoia have occasionally failed with some remote environment setup.

Examples: [2/4](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/51/), [1/31](https://drake-jenkins.csail.mit.edu/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/46/).

### What operating system are you using?

macOS 15 (Sequoia)

### Relevant log output

```shell
-- Testing download of remote cache server: 'http://10.221.188.9/'
dyld[842]: file does not start with MH_MAGIC[_64]: 0x00000000 0x00000000
```

tyler-yankee, 2025-02-05:
[2/5](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/52/)

tyler-yankee, 2025-02-06:
[2/6 cmake](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/55/), [2/6 bazel](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-bazel-nightly-everything-release/49/).

tyler-yankee, 2025-02-07:
This is probably occurring due to some weird error with the VM giving bad bytes for `curl`, which is being used for the CMake file download. See [#304](https://github.com/RobotLocomotion/drake-ci/pull/304) for the change to remove this section altogether.

BetsyMcPhail, 2025-02-10:
2/10 https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-sequoia-clang-bazel-continuous-release/210

tyler-yankee, 2025-02-11:
[2/11](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/62/)

BetsyMcPhail, 2025-02-12:
2/12 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/64

BetsyMcPhail, 2025-02-13:
2/13 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/63

tyler-yankee, 2025-02-18:
I did some digging into what `dyld` is doing with `ctest` under the hood. See the flags turned on here, from the top-level wrapper script: [branch on drake-ci](https://github.com/RobotLocomotion/drake-ci/tree/macos-link-debug). Looking at the logs between a [success](https://github.com/user-attachments/files/18854132/10.txt) and a [failure](https://github.com/user-attachments/files/18854133/11.txt), there doesn't seem to be a discernible difference with regard to the architectures of the libraries (especially looking at `libcurl`) being linked.

I also went into MacStadium and took a look at what `ctest` was linking against: `otool -L /opt/homebrew/Cellar/cmake/3.31.5/bin/[cmake|ctest]`. Nothing looked out of place to me in terms of needing to target `arm64` and/or `arm64-macos`.

This is after @williamjallen rebuilt the Sequoia images as well, so I'm not sure what else should be done here. This error is very elusive...

jwnimmer-tri, 2025-02-18:
Did we decide yet if its correlated with a specific physical machine?

tyler-yankee, 2025-02-19:
I can monitor that when running experimental builds manually and see if I notice a pattern. Ideally we'd add it to the logs, but extracting the hostname of the physical machine from inside the VM is nontrivial.

tyler-yankee, 2025-02-21:
2/21: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/74/

BetsyMcPhail, 2025-02-28:
2/28: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-bazel-nightly-everything-release/73

jwnimmer-tri, 2025-03-03:
This is happening way too often, causing a lot of buildcop pain.  We rarely have a green-slate nightly build anymore.  Bumping up to Kitware's highest priority now.

tyler-yankee, 2025-03-03:
We have a ticket open with MacStadium that they are looking into. Still unable to track to a specific machine. The node upgrade from macOS 15.0 to 15.3, which they said may resolve issues with the Virtualization Framework, did not affect anything.

We'll keep poking at it and see if we can track down any patterns.

tyler-yankee, 2025-03-03:
3/3: 

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/83/
* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-bazel-nightly-everything-release/76/

jwnimmer-tri, 2025-03-03:
(1) Can we make an accelerated reproducer?  So far, no luck.

(2) For the failing command, can we add a retry?  Print out the hexdump of the first 1K of the file?  Can we dump the md5sum of the program?

tyler-yankee, 2025-03-04:
3/4: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-bazel-nightly-everything-release/77, https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/84/

tyler-yankee, 2025-03-05:
> For the failing command, can we add a retry?

FWIW, we are calling `ctest_submit` with `RETRY_COUNT 4 RETRY_DELAY 15`, so it seems that this `dyld` error so fatal that it supersedes the retry attempt.

Still looking into this more today.

tyler-yankee, 2025-03-05:
It's possible that this is related to #20718. I'm going to test this more and follow-up; otherwise, nothing else stands out at this point.

Worth noting that we started observing this issue 1/31, which is the date we rebuilt both Sonoma and Sequoia unprovisioned images and the Sequoia provisioned images. Perhaps related to the changes in #22477?

Also worth noting that this issue has only been observed on the Sequoia provisioned images.

tyler-yankee, 2025-03-06:
Yes, that is more along the lines of what I meant. We rebuilt images on 2/8, but perhaps that wasn't enough time for a new version of some package causing the error to come out and be fixed; perhaps worth going again.

Turns out it's unrelated to #20718, since it still occurs after implementing the workaround for that.

ggould256, 2025-03-06:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/88/console

tyler-yankee, 2025-03-08:
Rebuilt Mac Sequoia provisioned image on 03/08. If it was caused by some upstream package issue, this should hopefully resolve it. Will continue to monitor.

ggould-tri, 2025-03-11:
* https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sequoia-clang-cmake-nightly-everything-release/93/console `dyld[803]: file does not start with MH_MAGIC[_64]: 0x00000000 0x00000000`
* https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-sequoia-clang-bazel-continuous-release/298/console `dyld[825]: file does not start with MH_MAGIC[_64]: 0x00000000 0x00000000`

BetsyMcPhail, 2025-03-12:
3/12 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/93/

SeanCurtis-TRI, 2025-03-14:
3/14 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-cmake-nightly-release/95/

SeanCurtis-TRI, 2025-03-18:
3/18 https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-sequoia-clang-bazel-nightly-everything-release/92/

tyler-yankee, 2025-03-18:
@BetsyMcPhail, @williamjallen, @jwnimmer-tri: I believe we have a fix (even if just for now) on this. I installed [CMake 3.31.6](https://github.com/Kitware/CMake/releases/tag/v3.31.6) manually on the Sequoia provisioned image, as opposed to using Homebrew CMake. (FWIW, I was also able to make an accelerated reproducer by running through our entire suite of CI scripts, but only executing a single Bazel test as opposed to `//...`). 

At this point it's clear that the error is somewhere in the virtualization layer in MacStadium's cloud. As such, I think this installation of CMake on the provisioned image is sufficient. (Especially if we're considering moving away from this soon.)

One minor note here: doing this requires adding `/Applications/CMake.app/Contents/bin` to `PATH`. I've already done that on the image that's currently saved, but would it be safe to do that [here](https://github.com/RobotLocomotion/drake-ci/blob/b6046eee4aa258240303f46b7a32451c1df0f8d9/ctest_driver_script_wrapper.bash#L48) as well?

tyler-yankee, 2025-03-19:
Rebuilt Sequoia provisioned image on 2025-03-19 with a symlink from `/Applications/CMake.app/Contents/bin/[cmake|ctest|ccmake|cmake-gui|cpack` to their respective locations in `/opt/homebrew/bin/` on the Sequoia image itself. If this ends up being the fix (and if we stay with MacStadium), we'll consider a better long-term solution.

jwnimmer-tri, 2025-03-24:
If we don't see this again in the next ~1 week, we'll close this as "seems okay now".

closed 2025-03-30 by jwnimmer-tri